nightscout/AndroidAPS #805: [AAPS3.0_beta2] Incorrect insulin amt shown after new pod activated.

tjbain, 2021-10-30:
Slow, but gradual beeping sounds after I deactivated a  DASH pod. Started a new pod, but beeping continued. Several screens showed 50u, despite the pod being full (pics attached). Force closing the app didn't solve the problem. Beeping stopped once I deactivated the new pod, but screen shows "unknown pod failure". Everything happened starting around 2:45am.

New Pod works like normal, but screens still show 50u in yellow, despite having a full pod.

AAPS3.0 Beta2
G6 CGM
Dash pod

[AndroidAPS_LOG_1635501672753.log.zip](https://github.com/nightscout/AndroidAPS/files/7445504/AndroidAPS_LOG_1635501672753.log.zip)
[image]
[image]
[image]

vanelsberg, 2021-10-30:
The amount showed is correct.
It is the amount of insulin needed (as reported by the Pod) to activate. Includes the 0.3U to prime the canula on canula insertion.

MilosKozak, 2021-10-30:
Ominipod cannot report reservoir value above 50U

closed 2021-10-30 by MilosKozak