nightscout/AndroidAPS #806: [AAPS3.0_beta2] Dash screen showing poor connection quality

tjbain, 2021-10-30:
Perhaps related to the previous issue I submitted, but after 12 hours with the new DASH pod, I'm showing poor connection quality (18%). As far as I know, I haven't experienced any issues with dropped connections, but the red makes me it seem like this is concerning. Logs and screenshot attached. I have not been far from my phone today, so I'm not sure why there'd be connection issues. I did not have any similar problem with Beta1.

AAPS3.0_beta2
Dexcom G6
Omnipod DASH
Samsung Galaxy Note 20 5G phone

[AndroidAPS_LOG_1635555626654.log.zip](https://github.com/nightscout/AndroidAPS/files/7445516/AndroidAPS_LOG_1635555626654.log.zip)

[image]
.

avereha, 2021-10-30:
This is what I wrote in the testing channel some time ago:

I think the only actionable point for this number is this: if it's low all the time, then we should mark the phone as 'not supported'. If it's low(<50%) with a specific pod, don't open any issue about other errors because the pod is probably broken.

A few details about 'connection quality' and how to use it:
  - This field and 'delivery status' are displayed only if engineering_mode is enabled. Regular users will not see this. That's why I don't think it's very important to make it very accurate(see https://github.com/0pen-dash/AndroidAPS/issues/95)
 - the number represents the percentage of successful connections: (number of ok connections/number of tries) * 100
 - A lot of things can make this number go down, from moving away from the phone, to bad lot of pods, where do you put the pod, sleeping on the pod, etc. 
 - if it's low only with some pods, again, it could mean it's a bad pod and not worth filling issue about this, because the pod is broken anyway(see for example: https://github.com/0pen-dash/AndroidAPS/issues/85)
 - in practice, I noticed that for me it is usually at 80/90%, but it can get as low as 50% sometimes(when I use a pod on the leg and keep the phone on the table, move away from the phone), etc. I use Samsung S10.
 - again, if this number goes low from time to time this is totally expected, we are more interested in the cases when it's consistently low

vanelsberg, 2021-10-30:
Confirming on @avereha

closed 2021-10-30 by MilosKozak