nightscout/AndroidAPS #807: [AAPS3.0_beta2] NSClient - xhr poll error

old-square-eyes, 2021-10-30:
Just installed beta 2 of NSClient 3.0. To get the build to work on my Mac M1, [I had to add a new version of "Room" to the gradle build file.
](https://stackoverflow.com/questions/68922875/room-database-is-not-working-in-mac-book-pro-m1). Not sure if that's relevant.

NSClient opens on my S20 fine (had to uninstall prev version for it to install). I completed the wizard. Lots of errors in the logs.

Now my NS site has gone down, so this seems pretty nasty.

[image]

[image]

old-square-eyes, 2021-10-30:
Phew. I uninstalled NSClient, and reloaded the Nightscout Heroku Dynos and Nightscout came back.

old-square-eyes, 2021-10-30:
Reinstalled NSClient and completed Wizard without NS creds. Then manually entered, and this time it started syncing status/BG. But eventually crashed the NS site again. Perhaps it's a internal server error at NS because it can't keep up with the sync?

old-square-eyes, 2021-10-30:
Finally seems to be syncing. I turned off my two OpenAPS rigs and the NS API was able to keep up.

I see data in the history viewer. But not in the main graph. I'll see if there is more syncing. But I also can't select a Profile per https://github.com/nightscout/AndroidAPS/issues/808

old-square-eyes, 2021-10-30:
I gave up. No plot on main graph. I went back to the old 2.X version and the plot immediately appeared.

MilosKozak, 2021-10-30:
this is not enough info to be able help you

MilosKozak, 2021-10-30:
where do you host NS?

old-square-eyes, 2021-10-30:
Heroku / Atlas Mongo

old-square-eyes, 2021-10-30:
@MilosKozak would you consider enabling local BG collection from xDrip? It's something I have wanted anyway to keep the watch face from NS lag.

savek-cc, 2021-11-01:
@old-square-eyes did the event happen when adjusting the phone time for daylight savings time? (I saw the same errors but thought it might be due to altering time and timezone in anticipation for the time rewinding 1h (combo pump).

old-square-eyes, 2021-11-01:
@savek-cc, I wasn't adjusting TZ, no. It's almost certainly a Heroku instance /dyno size. The API gets exhausted.

betees82, 2021-11-02:
I faced the same problem. Finally downgraded to NS client 2.8 and works ok. I'm not sure, but it might been related with profile switch.
[image]

closed 2021-11-04 by MilosKozak

old-square-eyes, 2021-11-06:
@MilosKozak is this in beta 3? I see no such option and a fresh install of beta 3 and I'm still getting the error. Also you mention NS upload. I'm pretty sure this is for data download. Sorry if I am stating the obvious.

under-control, 2021-11-06:
same problem here, it worked like charm till yesterday, restarting does not help
and I'm using build for MedLink (instead of RileyLink)

[image]

under-control, 2021-11-06:
Turns out it is caused by internet connection.

When connected to WiFi errors occurs, but then when switched to mobile Internet it connects right away and sends the data to NSClient.

old-square-eyes, 2021-11-06:
That's interesting. I wonder if mobile data naturally has a slower request speed so NS doesn't get exhausted. I'll test. My WiFi is quite fast.

old-square-eyes, 2021-11-06:
Fresh install / mobile data. It ran OK for a few minutes then same error for me.

old-square-eyes, 2021-11-06:
Still having the no profile error too.

MilosKozak, 2021-11-06:
OAPS doesn't support effective profile switches. you cannot make it working

old-square-eyes, 2021-11-06:
@MilosKozak I think you may have misunderstood. I'm just trying to get NSclient working. Nothing to do with OpenAPS

MilosKozak, 2021-11-06:
to activate profile you need data from AAPS
at least until there will be effective profie switch support in NS

MilosKozak, 2021-11-06:
NSClient cannot create effective profile switch
but pumpcontrol could

old-square-eyes, 2021-11-07:
@MilosKozak, I see. Well maybe I am mistaken. I assumed that my inability to select or otherwise apply a profile was the reason for no BG plot / NSClient IOB etc. Perhaps the issue is something else. Why then do I have no plot on the main screen? Per my screenshot and video above?

under-control, 2021-11-07:
I think I found the source of problem and it's not just WiFi itself.

It was xiaomi setting that I turned off to not to restrict data usage, but it has to be ticked to work connect properly with NS.

[image]

old-square-eyes, 2021-11-07:
@under-control I'm on samsung and have the issue with WiFi and mobile data. I'm 99% sure it's a rate limit / server error due to too many reasons. Whatever setting you're using must simply be attenuation the rate or coincidentally working.

old-square-eyes, 2021-11-09:
Confirming this seems fixed in beta 5. Rate limit seems to be doing its thing.

paulosnsilva, 2022-09-10:
I'm having this issue. I'm using AAPS v. 3.0.0.1 with NS v.14.2.5 hosted on 10be.de.
On AAPS I have the "slowdown uploads" options active.
I've tried several restarts on NS and on my phone too.
I use NS to gather data and create medical reports for my doctor.

paulosnsilva, 2022-09-10:
I've updated my AAPS to latest 3.1.0.3, but the issue persists.
Error: ERROR: io.socket.engineio.client.EngineIOException: xhr poll error.
I have 74896 pending records to submit. Maybe that is too much ? How do I reset it?

paulosnsilva, 2022-09-16:
Just for future record, I got it to work again by resetting my NS db, and deleting all my data from AAPS, After clean install, it started to work as expected. Maybe my queue got a bit too fat.