nightscout/AndroidAPS #808: [AAPS3.0_beta2] NSClient - Cant select NS Profile with 8hr DIA

old-square-eyes, 2021-10-30:
NSClient but probably also AAPS... I can't select my profile. Error below...

[image]

old-square-eyes, 2021-10-30:
I changed the DIA to 7 and now I can select it. But it doesn't apply. I tried creating a new manual profile in NS, and NSClient can see it, and select it but I get the same. It doesn't apply.

old-square-eyes, 2021-10-30:
[image]

And...

[image]

Alamo04, 2021-10-30:
In v3.0 only local profiles are allowed. 

You can download NS profiles into your local profiles and sync them, but using is only allowed for local profiles.

Perhaps that's the problem ?

old-square-eyes, 2021-10-30:
@Alamo04 - OK I'll try again. Hopefully that improves #807 and no plot too :)

MilosKozak, 2021-10-30:
MAX DIA if limited to 7hours

MilosKozak, 2021-10-30:
buton text is fixed by https://github.com/nightscout/AndroidAPS/commit/3a33529307da30810ae273123cda12cb0f1743f6

MilosKozak, 2021-10-30:
there is profile functional change. see release notes

closed 2021-10-30 by MilosKozak

old-square-eyes, 2021-10-30:
@Alamo04 I made @MilosKozak change https://github.com/nightscout/AndroidAPS/commit/3a33529307da30810ae273123cda12cb0f1743f6 and now I don't seem to have the ability to access my NS profile at all. Not even to make a local copy like you suggested. Though I'm still battling sync issues :(

old-square-eyes, 2021-10-30:
@MilosKozak can we please reopen this. I will rename it.

1. My NS sync finally complete
2. I couldn't download my NS OpenAPS profile so I created a manual, local one.
3. I saved, and activated the profile but NSClient still says "NO PROFILE SET"

I still have no data on my plot (but it's in the history browser)

old-square-eyes, 2021-10-30:
Here is a video...

https://user-images.githubusercontent.com/25760491/139558488-da2a9849-dbb5-45d9-bb88-0e0f403ed91a.mov

Alamo04, 2021-10-31:
Don't know what you have done, but it seems to be a user problem.

You can try to do a fresh installation:
- save your preferences
- deinstallation of AAPS 
- restart phone
- New installation of AAPS
- restart phone
- load your preferences
- control all settings that they are correct

Didn't have more ideas.

Alamo04, 2021-10-31:
Check all NightScout client settings, what you want to use.

[image]

Alamo04, 2021-10-31:
Profile creation and set in AAPS.
[image]

[image]

bchris21, 2021-10-31:
> I changed the DIA to 7 and now I can select it. But it doesn't apply. I tried creating a new manual profile in NS, and NSClient can see it, and select it but I get the same. It doesn't apply.

I had the same issue when installed the Beta2. You have to change at least one basal in order for the profile to be applied. Once applied then change it back to the normal one.

Hope it helps.

Alamo04, 2021-10-31:
You can also read this tread, with that problem:
https://github.com/nightscout/AndroidAPS/issues/771#issuecomment-955699907

old-square-eyes, 2021-11-06:
@Alamo04 and @Philoul Changing % as suggested in #771 still does not set a profile. I am on beta 3 now.

All I see on the main page is "PROFILE NOT SET". I have set a manual local profile. And I even deleted at basal segment so it's not the same as before.

old-square-eyes, 2021-11-06:
@MilosKozak can we please reopen. I'll update the tile to be beta 3.

old-square-eyes, 2021-11-06:
@MilosKozak I'm getting a reproducible crash using the "copy profile" button.

MilosKozak, 2021-11-06:
the create new issue for that