nightscout/AndroidAPS #809: [AAPS3.0_beta2] NSClient - Cant silence alarm while sync in progress

old-square-eyes, 2021-10-30:
While setting up fresh and (still, presumably) syncing NS data, the main graph doesn't display, not do any of the info pills/widgets or or profile. Basically most features don't seem to work.

I got a loud "last loop" 30 minutes ago alarm because I walked away from my OpenAPS rig.

While NSClient is in this state there is no way to snooze or dismiss this alarm. I had to turn my phone off until I had returned to my rig and another loop completed.

MilosKozak, 2021-10-30:
what phone do you have? full NS sync is pretty CPU consuming

old-square-eyes, 2021-10-30:
Galaxy S20.

MilosKozak, 2021-11-02:
probably because of busy bus
hopefully this helps https://github.com/nightscout/AndroidAPS/commit/ff1a9b24dbd3a36361171aa85ff8eed54f8cb003

closed 2021-11-02 by MilosKozak