nightscout/AndroidAPS #815: V3B2 : Bug data from different pomps

robertrub, 2021-10-30:
I deactivated the ending EROS and activated a new EROS. Process between 9:12 till 19:17. 
I got this error message once, validated and am continuing using the EROS pod normally till now.
Before deactivating EROS, I had activated a DASH with the same phone in parallel. I'm not sure if they are related.

[image]

samspycher, 2021-11-01:
Hi Rob!

Thanks for reporting this. It is a duplicate of #782 and #775.

May I ask: did you experience any freezing of the screen while changing the pods?

robertrub, 2021-11-01:
Hi @samspycher 

I was waiting for the freeze but no, nothing abnormal happened. 

I thought it had frozen as nothing was moving on the screen but I waited patiently and the next activation phase arrived without any intervention. I didn't get the Android "AAPS is not answering, want to close or wait ?" message. I think the process didn't took longer than usual but I'm not sure.

robertrub, 2021-11-02:
Hi again @samspycher 

I changed my EROS pod just now (with the latest dev 258224055) and I had the freeze when I put the pod on my arm and hit the OK on the little popup message. All went OK with the EROS but I got the Android message "Wait or Close". I hit wait and AAPS came back to life and the pod is working OK. Then, I got the same "data coming from another pump" error message.

Attached is the logs. Freeze and message between 14:17 and 14: 20.

[AndroidAPS_LOG_1635859291848.log.zip](https://github.com/nightscout/AndroidAPS/files/7461010/AndroidAPS_LOG_1635859291848.log.zip)

MilosKozak, 2021-11-02:
what did you click before the freeze?

robertrub, 2021-11-02:
I hit the OK button, there

[image]

and the "popup" disappeared, I was back to this page

[image]

avereha, 2021-11-13:
I think we can close this and reopen if you still see the issue in dev/next beta.

The 'data from different pumps' should be fixed with https://github.com/nightscout/AndroidAPS/commit/85178ee19ff628dfe4422409973c6f8ae7b511b5

'screen freeze' should be fixed with https://github.com/nightscout/AndroidAPS/commit/52e343213249c0abff26aab4756b5c9edf8755c3

closed 2021-11-13 by robertrub