nightscout/AndroidAPS #818: When on DASH history view, AAPS crash/restart when rotating the phone

vanelsberg, 2021-10-30:
**AndroidAPS 2.8.2.8-dev**
Build: 8fe7e7ce2
Flavor: fulldebug

**Problem description:**
When on DASH history view,
AAPS crash/restarts when rotating between portrait<->landscape mode (rotating the phone)

kristinsmotz, 2021-10-31:
Can confirm this happens for us too
-----------------------------------------------------
AndroidAPS 3.0-beta2
Build: f49c93f22-2021.10.30-15:01
Remote: https://github.com/nightscout/AndroidAPS
Flavor: fullrelease
Nightscout version: 14.2.2
Engineering mode enabled

Philoul, 2021-10-31:
@avereha I confirm crash with kotlin version
I tried with first java version and it works, but I couldn't find in kotlin what's wrong...
=> That's what I have in logcat
`2021-10-31 17:42:29.748 5559-5559/info.nightscout.androidaps E/AndroidRuntime: FATAL EXCEPTION: main
    Process: info.nightscout.androidaps, PID: 5559
    java.lang.NullPointerException: Parameter specified as non-null is null: method kotlin.jvm.internal.Intrinsics.checkNotNullParameter, parameter view
        at info.nightscout.androidaps.plugins.pump.omnipod.dash.ui.DashPodHistoryActivity$onCreate$3$1.onItemSelected(Unknown Source:2)
        at android.widget.AdapterView.fireOnSelected(AdapterView.java:957)
        at android.widget.AdapterView.dispatchOnItemSelected(AdapterView.java:946)
        at android.widget.AdapterView.selectionChanged(AdapterView.java:935)
        at android.widget.AdapterView.checkSelectionChanged(AdapterView.java:1118)
        at android.widget.AdapterView.handleDataChanged(AdapterView.java:1094)
        at android.widget.AbsSpinner.onMeasure(AbsSpinner.java:193)
        at android.widget.Spinner.onMeasure(Spinner.java:614)`

vanelsberg, 2021-10-31:
@Philoul @avereha This AAPS had this problem (early 2.8.1?) on the main overview. Took some time to fix that.
Not sure how and by whom it was solved. Maybe @MilosKozak knows?

MilosKozak, 2021-10-31:
https://github.com/nightscout/AndroidAPS/commit/a7ce4c0150627dbe85604a401067966d789ca028

vanelsberg, 2021-11-01:
> [a7ce4c0](https://github.com/nightscout/AndroidAPS/commit/a7ce4c0150627dbe85604a401067966d789ca028)

@MilosKozak 
Confirming: problem solved. Tnx!

closed 2021-11-01 by MilosKozak