nightscout/AndroidAPS #819: Profile Activation may fail due to profile error handling issues

rustymonkey, 2021-10-31:
When importing settings from 2.8.x or creating a new profile from scratch there appear to be some issues regarding how missing/out-of-range values are handled:

1) Imported (or newly created) profiles that appear to be valid, and can be activated in the 'Profile' tab, appear to get activated, but are shown as inactive on the Overview screen. 

To replicate:
1 import settings (or delete all existing profiles and create a new profile) containing a local profile with a target of 5.4 to 5.7 mmol/l
2 on the Profile tab, edit an ISF value from 4.5 to 4.6 and save the profile
--> No errors result
3 Activate the profile
--> The activation confirmation dialog appears and it appears to activate
4 Go to the Overview tab
--> It indicates in red 'NO PROFILE SET'!
5 on the Profile tab, change the lower end target from 5.4 to 5.5 and save it
6 Activate the profile
--> This time it correctly activates

So there are two aspects this issue:
a) a target below 5.5 is invalid, but upon saving the profile no error is generated so the user thinks that the profile is valid
b) Upon attempting to activate the invalid profile, no error is generated so the user thinks that it activates successfully

note: I haven't tested other variables besides 'Target' to see if this issue occurs with them

2) When creating a new profile, out-of-range errors often appear confusingly or incorrectly:
a) Confusingly for items other than the value that was just configured. e.g. if the user configures DIA, then an error appears for IC. The design rationale was presumably to let the user know that IC isn't yet configured, but the implementation is confusing because the error has nothing to do with the just-completed action.
b) Incorrectly for items that were previously configured correctly. e.g. if a user configures ISF, an error regarding incorrectly configured IC will often display even when IC was configured correctly.

To replicate:
1 Delete all profiles and create a new one
2 Set DIA to 5.5
--> '! Error in IC values'. This is unexpected because the user hasn't yet configured an IC value.
3 Set IC for 12am to 5.5
--> '! Error in IC values' displays! This is unexpected because a valid IC value has been entered
--> After about a minute, the error changes to '! Error in ISF values' (confusing because the user hasn't entered ISF values)
4 Set ISF to 5.0
--> Similar issues as at step 3 (except this time ISF and Basal errors)
5 Set Basal to .70
--> 'Error in IC values!!' Why is there an error in IC values now?!
--> This error then disappears (and doesn't indicate anything about missing 'Target values')
...

A possible solution would be to get rid of these error messages entirely, and just highlight in red any heading containing missing/invalid settings (DIA / IC / ISF / BAS / TARG). And the only time that an error message should appear is to clarify on the active tab why that particular header is in red. e.g. if the user clicks the red Target tab 'Error: target BG must be at least 5.5'. That way users would know _all_ of the items that remain to be correctly configured, and the problems of errors appearing confusingly/at the wrong time will go away.

-----------------------------
Tested on 3.0 beta 2

rustymonkey, 2021-10-31:
Issue 1) is similar/related to: #630 #771 #808

One more piece of information: issue 1) seems to have stopped occurring now that I've managed to correctly set up a profile. Even more strangely, the profile has a lower Target of 5.3 and AAPS is no longer rejecting it! So I'm no longer certain that the target setting below 5.5 is the sole root cause. 

The only thing that I do know is that the issue can be replicated via uninstall/re-install AAPS and re-import 2.8.2 settings. I now suspect that another aspect to the issue might be that 'Activate profile' may, in that specific situation (import after clean install ?with out of range target?), fail to trigger pump communication on profile change. In my case, I've found that clicking 'Activate profile' (even after changing a profile setting) didn't trigger any communication with the pump nor generate an error message about failure to communicate with the pump. Clicking 'Refresh' on the 'Medtronic' tab initiated a pump wakeup and profile update.

I've submitted logs of this.

---------------------------
Note: tested with 3.0 beta 2 wl MDT 554 / RileyLink / PixelExperience(Android 11)

MilosKozak, 2021-11-02:
1) resolved by https://github.com/nightscout/AndroidAPS/commit/760261e258e51e0d8f290e1ae9322df7bc1e0ff1
2) this is expected behavior. The goal is tell to the user where is a mistake in profile if background is red

open new issue if you find something new

closed 2021-11-02 by MilosKozak

rustymonkey, 2021-11-02:
1. Thanks! I'll retest the next build.
2. Although this doesn't prevent AAPS from working, the current behavior isn't expected **by the user** because:
a) Errors are unrelated to the activity that triggers them. Users will eventually figure it out, but it would be preferable to show the status of unconfigured items in advance rather than in response to configuring an unrelated UI element.
b) Errors persist (e.g. for a minute) even after a valid profile has been configured and enabled. This is more serious because it temporarily gives the impression that a valid profile is faulty.

A simpler solution than the original suggestion might be to display the live (i.e. no timer) error status for **all** configuration values associated with the current profile on all tabs of the profile screen. e.g.
! Invalid profile: <ProfileName>
! Unconfigured: Basal
! Out of range: ISF, Target

MilosKozak, 2021-11-02:
2b is fixed in beta3. it's related to android toast duration

MilosKozak, 2021-11-02:
ad 2a) messages are in order of tabs on screen ie. if you fix DIA, error message for IC is displayed (if wrong) etc

rustymonkey, 2021-11-04:
Tested Beta 3 and verified the fixes to the functional problems. A couple of minor usability issues remain:

1b) If the user tries to 'Activate Profile', but AAPS fails to do so, there's still no feedback to the user on the profile page. An example of this is when issue #843 occurs (though I expect that there are other cases when this can occur as well). The user can see on the Overview tab that the profile failed to activate, but there's no error/feedback on the Profile tab about what the problem might be.

2a) Though I understand what the app is doing, it's still confusing to a new user when they click to adjust e.g. the IC value, an error about the ISF appears.