nightscout/AndroidAPS #822: [3.0beta2] Graph render glitch when changing between 6/12/18/24 hr view

kristinsmotz, 2021-10-31:
-----------------------------------------------------
AndroidAPS 3.0-beta2
Build: f49c93f22-2021.10.30-15:01
Remote: https://github.com/nightscout/AndroidAPS
Flavor: fullrelease
Nightscout version: 14.2.2
Engineering mode enabled

Noticed this on both NSClient and AAPS phone. When long pressing the graph area to change between views, the graph briefly displays the requested number of hours of history, before rendering predictions and shortening the history part of the view. When the history is shortened, all "glucose dots" seem to pile up in the left-most position causing a vertical line as high/low as glucose had been during the 2-3 hours before the first visible time. When next glucose value is received, the graph renders correctly again. 

[image]

[image]

MilosKozak, 2021-11-01:
probably caused by DST change

kristinsmotz, 2021-11-01:
I don't think so, it has done that since beta 1 for me, and still does it today.

kristinsmotz, 2021-11-01:
From nsclient
https://user-images.githubusercontent.com/71758682/139753255-effe3115-2213-456b-9aaf-af289674dd57.mp4

From AAPS phone 
https://user-images.githubusercontent.com/71758682/139753272-f2d32bb4-95e2-4b1b-9ebf-64f7b42095a6.mp4

MilosKozak, 2021-11-02:
BG should be displayed immediately
the delay of displaying the rest of the data is expected because they must be calculated first

or is there something else wrong?

kristinsmotz, 2021-11-02:
The glitch I am talking about is the bg's (dots) for the time period being removed from the left side of the graph when the prediction renders on the right side, the dots pile up in a vertical line on the left side. It's most visible when there's been a high variation of BG's in the time period just at the start of the display period window. (4-6 hours ago in 6hr view, 10-12 hours ago in 12hr view, etc). At first, the graph will display the past 6 or 12 hours for a second or two, then render predictions for the next 2-3 hours, at which time it removes the earliest part of "history" from the graph, so that you see 2 hours of future and 4 or 10 hours of past history. At that time, the historical BG's "pile up" in a vertical line. When BG syncs, OAPS runs, and new predictions are rendered, the vertical line disappears. In this image I have circled the vertical line/pile-up. 

[image]

Of course this is a very minor glitch, not anything affecting functionality, just a minor annoyance that I hadn't seen in the previous version (2.8.2), so I thought I'd bring it to your attention in case there's also an easy fix.

MilosKozak, 2021-11-02:
i cannot replicate the issue. pm me your ns address

kristinsmotz, 2021-11-02:
Sure I'll do that. Just wanted to add that this glitch is only visible after long-pressing the graph area to switch views, and waiting for the predictions to render. Then when glucose is next refreshed, it disappears again.

Philoul, 2021-11-02:
@MilosKozak I confirm (this issue is similar to #801, you can see the vertical bar on my second screenshot).
I tried to fix it a few months ago with always preparing data for max range of data for all curves (24h + prediction) and  return selected range.
I made a PR (not merged) that fixed the temporary vertical bar on the left and displayed immediatly all curves on rescale, but I closed it because it was not compatible with History Browser 😞...
My other idea was to only calculate required data on new entry or each 5 min (most of time 99% of graph values are already calculated, and it's only when we enter insulin or carbs in the past that we have to calculate again a subset of values), keep in memory 24 or 48h of graph values and save older one in database to be able to show immediatly all curves in historybrowser (without calculation) and in overview...

MilosKozak, 2021-11-02:
resolved by https://github.com/nightscout/AndroidAPS/commit/2ad7d9f4d236ca6adb76b6afa9e51ecd3a354bb4

closed 2021-11-02 by MilosKozak