nightscout/AndroidAPS #823: 3.0 beta2 Statusscreen stopped updating insulin amount

RalfCGM, 2021-11-01:
3.0 beta 2

Happened since 30 okt, don't know exact time. 

Statisticus screen does notr "updating" when selecting, nut doesn't show the right insulin amounts. Status for TIR do seem to be right. 

Somehow, since updating the log files aren't updated anymore, so can't provide logs. 

[image]

Reporting bugs

RalfCGM, 2021-11-01:
Now, after a reboot of the phone, or after some time (dont know what caused it) it looks like giving the right values. 
[image]

MilosKozak, 2021-11-01:
If it works today it's cased by DST change

RalfCGM, 2021-11-02:
Ive changed the time on my phone om the 30st, not the 31st. But  it's working today, so you might be right. Ill close this issue.

closed 2021-11-02 by RalfCGM