nightscout/AndroidAPS #827: V3B3 EROS / Basal settings twice to pump

robertrub, 2021-11-01:
This happened several times but can't duplicate yet.

I have an automation that switches the profile to 120% when I have high BGs or high positive deltas. Sometimes the basal rates are sent twice to the EROS pod. 

Maybe (not tested yet) it has to do with me "Okaying" the basal change in AAPS as soon as it appears.

[image]

[image]

I'll attach logs. Strangely I need to edit the message in order to be able to attach a file...

robertrub, 2021-11-01:
One more exemple. In one (1) a normal switch to 120% and back (theoretically *10* minutes later...) and then going back to 120% once more as the high BGs were still there. The 2nd change was sent twice.

I'm sure that at 0 AM, I was sleeping and didn't validate the message in AAPS. Thus, my first supposition above is wrong.

[image]

[image]

robertrub, 2021-11-01:
I've just did a **_manual_** profile change to 130% for 20 min and it got sent twice to the pump. Maybe it is not the automations that it the origin of the bug. Still testing...

kristinsmotz, 2021-11-02:
I believe I have also seen this with both manual and automation profile changes, using Omnipod Dash with 3.0beta2. Don't have screenshots right now, will check when AAPS phone gets home from school.

robertrub, 2021-11-02:
@MilosKozak I cornered the bug :) I can duplicate 3 times in a row. I'm the best ;) ;) ;)

Long press the profile rectangle, set 120% for 20 min, validate. EROS will beep once and several seconds later, ANDROID will show a message "Basal profile set in the pump"

[image]

Right swipe the Android message and the basals will be sent again to the pump and the same message will appear again. 

If you swipe the message "Basal profile set in pump" the 2nd time also, it will not send the profile a third time to the pump.

I hope you have enough to corner the bug in the code too. You of course is better than best as I find the bug but you can fix it :)

robertrub, 2021-11-03:
Ok, it seems that I was too optimistic. It is NOT related to swiping  the Android message. It sends basal profile twice even when I don't swipe.

robertrub, 2021-11-05:
I just changed for on-body DASH and I see that DASH too sends basal settings twice.

[image]

kristinsmotz, 2021-11-05:
I managed to catch this as well with Dash.
On 3.0beta3.

[image]

vanelsberg, 2021-11-05:
**Confirming:**
Resetting a Temp profile to 100% is executed 2 times, both generation a notification.

**Repro:**
**1)** Set a Temp Profile for 120% (this will do one single Suspend/Set Basal).
**2)** Wait for the "Profile set OK" notification.
**3)** Execute a testbeep to mark the current time in history.
**4)** Cancel the Temp Profile by setting it back 100%.
**5)** Wait for the "Profile set OK" notification.
**6)** Execute a testbeep to mark the current time in history.
**7)** Wait for the _second_ "Profile set OK" notification.

I get the notification "Profile set OK" and then some time after a second "Profile set OK" notification.
Check history view and find two identical Suspend/SetBasel entries.

avereha, 2021-11-08:
I was not able to reproduce this with latest dev. Can you still reproduce it?

kristinsmotz, 2021-11-08:
Reproduced just now, on beta5 installed today.
[image]
[image]

vanelsberg, 2021-11-08:
@avereha 
> I was not able to reproduce this with latest dev. Can you still reproduce it?

Yes, Still repro om build commit#2b021ffaf
Try acknowledging the set profile by swiping the phones notification?

Just noticed most of the time the second "set basal" only happens when I immediately acknowledge the first one by swiping the Samsung S9 notification. (The snooze button stays ineffective for some time, but I my Samsung S9 I can swiftly acknowledge by swiping the notification)
When I wait until I can press the AAPS snooze button, this does not happen.

robertrub, 2021-11-08:
Yes, had it just a couple of hours ago. Last dev or maybe before last... Milos is making new version several times a day...

[image]

robertrub, 2021-11-08:
> Just noticed most of the time the second "set basal" only happens when I immediately acknowledge the first one by swiping the Samsung S9 notification. (The snooze button stays ineffective for some [...]

Nope, on my S9, it happens even when I don't touch the phone. I'd thought the swipe was the culprit but it is not.

avereha, 2021-11-09:
Will you please add some logs when you notice  his bug on Dash?
I tried to switch profiles >5 times, and could not reproduce this bug :(.

robertrub, 2021-11-10:
@avereha I cornered your bug :) :)
The bug arrived when I turned ON beeps for basal. If beeps for basal is OFF, the basal is sent once. ON, they are sent twice ;)

[image]

[image]

[image]

Logs arounf 6:35 AM

vanelsberg, 2021-11-10:
@robertrub 
> If beeps for basal is OFF, the basal is sent once.

Negative. I have "Basal beeps enabled" set to "off".
Getting the basal set twice....

vanelsberg, 2021-11-10:
[AndroidAPS_LOG_1636540448451.log.zip](https://github.com/nightscout/AndroidAPS/files/7511962/AndroidAPS_LOG_1636540448451.log.zip)

Some logging.
What I did:

1. Delete logfiles.
2. Beep
3. Profile set 120-100% (got 1 notification, pressed "snooze" button af about 1 min)
4. Beep
5. 100 -> 120% (got 1 notification, pressed "snooze" button af about 1 min)
6. Beep
7. 120 -> 100% (got 1 notification ("confirmed notification by swiping)
Got the second notification some seconds later ("confirmed notification by swiping)
8. Beep
9: Send logs

[image]

vanelsberg, 2021-11-10:
Btw: settings:
[image]

robertrub, 2021-11-10:
_Negative. I have "Basal beeps enabled" set to "off".
Getting the basal set twice...._

Every time I think that I've corned this one, it kicks back :(   Seems each phone has a different trigger ????

avereha, 2021-11-11:
Ok, new idea: can you reproduce this if `NSclient` is disabled?
I see something in logs regarding NS, right before the second notification.

robertrub, 2021-11-11:
@MilosKozak It seems that @avereha corneted this bug for real :)
It is related to NS. 

In AAPS, I pause NS, I change back and forth a couple of times the profile between 100% and 120%. There are no double sends to the pod.

As soon as I unpause NS in AAPS, basal is sent once more to the pod.

[image]

[image]

[image]

vanelsberg, 2021-11-11:
Confirming: behavior as reported. Good find!

When NSClient is pausing, when setting new profile percentage there is no longer the second set basal on DASH overview an no second notification.
When resuming NSClient, the second set basal with notification occurs/

robertrub, 2021-11-13:
@MilosKozak This NOT an Omnipod problem. The calibration is doubled too.

I initiated the calibration from AAPS and BYODA came forward for me to enter my blood test result. I entered and validated and it asked again. 

Treatments in AAPS logged it twice too. Didn't test with NS synchro posed yet.

[image]

MilosKozak, 2021-11-16:
fixed by https://github.com/nightscout/AndroidAPS/commit/f48dcc77e23d499ab0c1af8499ce9ce8ed574ba1

closed 2021-11-16 by MilosKozak