nightscout/AndroidAPS #828: NSClient 3.0-beta2 profile sync problem

shafiro, 2021-11-01:
NSClient 3.0-beta2
Build: f49c93f22-2021.10.30-14:48
Remote: https://github.com/nightscout/AndroidAPS
Flavor: nsclientrelease
Nightscout version: 14.2.4
Engineering mode enabled
Device: Xiaomi MI 9, Android 11, MIUI 12.5.1

1. Profile is synced correctly from AAPS to NS.
2. Profile is not synced correctly from NS to NSClient.
3. Wrong profile name is shown on NSClient main screen (top left), but correct profile name shows on profile switch box.
3. BG is displayed in mmol/l, despite profile details shows units defined as mg/dl.
4. 3-line menu->Profile: shows profile in RED with "Error in ISF Values" despite all profile values including ISF are shown correctly.
5. Preferences->General->Units: None is checked.
6. After checking mg/dl, main screen becomes distorted for a short period, than shows values as they should appear. Also 3-line menu->Profile, shows profile correctly with no error. After a few minutes, all is back to as described on section 3-5 above.

Had same problem with 3.0-beta1
Tried Force stopping NSClient, Clear all data via android settings, delete app folders, reinstall.
Tried disabling/enabling sync from both AAPS and NSClient.
Could not find the cause.

[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/7452171/AndroidAPS.log)

[image]
[image]
[image]
[image]

[image]
[image]
[image]

MilosKozak, 2021-11-01:
Unit settings is taken from AAPS preferences ... not from NS profile anymore

MilosKozak, 2021-11-01:
pm me your NS address

MilosKozak, 2021-11-02:
resolved in dev hopefully
you are running profile name Eitan which is not in profiles anymore
do a profile switch on aaps phone with existing profile

closed 2021-11-02 by MilosKozak