nightscout/AndroidAPS #829: [3.0-dev] missed BG reading alert on every app startup

savek-cc, 2021-11-01:
On each and every App startup, I get a local alert because of "Missed BG readings" (see screenshot).
Apparently the alert is created before the last BG value is even read from the DB?
Can be re-created by just stopping aaps and starting it again.

On my daugther's phone, this causes an audible alert if e.g. we restart the phone in the night due to Dexcom or Pump BT issues, thus more often than not waking her up.

Solution(?): AAPS should not be sending alerts on app startup before initializing the different subsystems - even if this alert is dismissed a couple seconds later, the alarm sound has already been played.
[image]

robertrub, 2021-11-01:
I get the same alert when I start AAPS too. It is not useful as it gives an inexact information to user (theoretically _exact_ as AAPS didn't get BGs for some time but it will get them as soon as it finishes starting ;)  ) .

MilosKozak, 2021-11-01:
improved by https://github.com/nightscout/AndroidAPS/commit/f700c22966c6edbdee5a60caeb123290ad6dbba6

closed 2021-11-01 by MilosKozak