nightscout/AndroidAPS #830: [2.8.2] [Medtronic] AndroidAPS forgets it just bolussed/issues double bolus

ArthurusDent, 2021-11-01:
AndroidAPS sometimes forgets it just bolussed (SMB) and also sometimes issues a double bolus (SMB) within a minute which might be a connected issue. When re-reading pump history, AndroidAPS doesn't know those bolusses were SMBs but thinks the user entered them on the pump.

I kind of hope this happens due to a corrupted database and uninstalling and re-installing would fix it but to make sure this isn't a serious bug I thought I should report the behaviour here.

It seems to me that quite often something triggers a re-run of the algorithm not just after a new bg value was received while AndroidAPS was already communicating with the pump and AndroidAPS then wants to send a different command to the pump. Maybe something is being screwed in those cases and a bolus occurs that isn't being recorded.

This has now happened about four times that I know of.

Let's give you some screenshots first:
[image]
[image]
[image]
[image]

Here's also a video of the pump history:

https://user-images.githubusercontent.com/6848629/139664309-fa4bf1e4-e8dc-4ad0-a0c2-b088323f3b69.mp4

Here are the logs:
[Log - double bolus.zip](https://github.com/nightscout/AndroidAPS/files/7452833/Log.-.double.bolus.zip)

System:
AndroidAPS 2.8.2 194549adc
Oneplus 3T Android 11 (LineageOS)
Medtronic 554/523 (happens on both)
RileyLink
Dexcom G6 over xdrip

andyrozman, 2021-11-04:
That driver thinks SMB is Corr (after history is read) will be fixed with https://github.com/nightscout/AndroidAPS/issues/798, we can look at double bolus issue then... But this is really weird... Unless we receive 2 boluses we wouldn't issue 2 of them, but what might happen is that first bolus needs longer to be issued (pump communication)... I will take a look at logs later, to try to determine what was problem... 

Is double bolus reproducible? Does it happen when certain steps are executed.

ArthurusDent, 2021-11-04:
Ok, good to know the first problem is already being worked on!

What just came to my mind is that I noticed this behaviour with the double bolus fairly recent. I've been using 2.8.2 beginning with January but only witnessed the first double bolus in August and since then it happened three or four times. 

It is not reproducible and seems completely random to me.

I have just reinstalled and reset the database today to hopefully fix the double bolus issue. If need be, I might be able to restore from a backup I created before reinstalling so I could provide some further data. The logs have been saved spearately, if you need another example of this happening.

RalfCGM, 2021-11-10:
Don't understand the technicals, but I also saw similar behaviour with Combo on 3.0 dev 4. Noticed a duplicate smb right after an other one was issued. Noticed after that that there seems some recalculation done, as described here. Don't know / understand if this could be the same issue, but looks similar to me.

https://www.facebook.com/groups/AndroidAPSUsers/posts/3132853100269419/

andyrozman, 2021-11-14:
Mhm. So this might not be a driver problem itself...

ArthurusDent, 2021-11-17:
Ok, I have something else to report that might be connected. Keep in mind that this is a fresh install of AndroidAPS 2.8.2. For some reason the Open Humans uploader doesn't work but that might be a different problem.

Here's what happened and what I have witnessed quite often:
The algorithm runs at 16:25 because I have just stopped my TT and want to return to the standard target that is set in my profile.
The algorithm says it wants to send the max basal rate and an SMB with 0,20 IE. 
I think the max basal rate actually got set.
No idea if that is the problem, but the SMB cannot be issued right away because the pump history is being read.
Looking at the screen a minute later (16:26) the SMB never has been delivered and isn't queued but the algorithm recalculated with the sam bg and said there was nothing to do.
At 16:28 the algorithm runs again because a new bg was entered and this time it delivers the SMB as calculated.

Does the algorithm at 16:26 think it just delivered an SMB (less than 2 minutes ago) although it didn't?
But why would the algorithm run again at 16:26 when there is no new bg?

[AndroidAPS recalculation no SMB.zip](https://github.com/nightscout/AndroidAPS/files/7556047/AndroidAPS.recalculation.no.SMB.zip)

savek-cc, 2021-11-17:
For me it's that (almost?) all SMB get displayed as regular bolus after an app restart - so the screen is full of 0.1ie readings on the graph even though all of these were the small blue triangles for SMB before (Combo on latest dev).
So force stop app, start it again, no more triangles, only standard bolus.

MilosKozak, 2021-11-17:
with transition 2.8 -> 3.0 more than 50% od code was rewritten. Looking for bugs in 2.8.2 is wasting of time (mostly) because the code is probably changed in 3.0
please test 3.0

savek-cc, 2021-11-18:
My issue is with 3.0 beta - will open another issue.

RalfCGM, 2021-12-17:
I think i notice the same behavior sometimes. Notice it through a vibrating combo. Combo vibrates on tbr. I would suspect combo maximum vibrates only  once every 5 minutes but i noticed just now on 3.0 beta 9: tbr - bolus - tbr - tbr. It looks like some loop is happening. Ill provide Logs in a minute.

parapenT1sta, 2022-06-11:
This seems not be related to Medtronic only, since I have Accu Check and Combo and I am facing the same issue but in threatments, I can see the duplicated is coming from NightScout, I'm going to disabled and check if it fixes. I also have xDrip synching with NS and that might be the issue.

[AndroidAPS_LOG_1654975994191.log.zip](https://github.com/nightscout/AndroidAPS/files/8884632/AndroidAPS_LOG_1654975994191.log.zip)

[image]

ArthurusDent, 2022-06-11:
Did your pumps actually issue two boli? On Medtronic I can see two boli in pump history (in AndroidAPS and directly on the pump).

parapenT1sta, 2022-06-11:
> 

No, not at all. The only thing is that the calculator is showing me that I need carbs, in that last example because it doubled the info of IOB is telling me that I need more 60carbs., is only affecting the calculations nothing more.

parapenT1sta, 2022-06-11:
> 

Do you have "Receive" options enable in AndroidAPS NSClient? did you try to disabled all of them?

ArthurusDent, 2022-06-11:
No, I have not and I don't think this issue is connected to Nightscout. 

It looks like you are seeing a different bug. Have you seen this one https://github.com/nightscout/AndroidAPS/issues/1768 ?

parapenT1sta, 2022-06-11:
> No, I have not and I don't think this issue is connected to Nightscout.
> 
> It looks like you are seeing a different bug. Have you seen this one #1768 ?

Sorry! you're right. :(

MilosKozak, 2023-08-17:
Closed for inactivity. Create new issue is if you still see the problem on latest branch

closed 2023-08-17 by MilosKozak