nightscout/AndroidAPS #831: V3B2 and last dev : Missing BG dots on the graph

robertrub, 2021-11-01:
It happened with V3B2 and in the last dev.
[image]

Sometimes, the left side of the graph has no BG points and then they appear next time I look at them... No idea what is the cause...

[image]
[image]

Philoul, 2021-11-02:
Do you see that after a long press on graph to change time range or at any moment (without doing anything else) ?

Alamo04, 2021-11-03:
Confirmed.

For me, it was fixed after a few long press to the graph until comming back to the wanted time range.

Alamo04, 2021-11-07:
This issue still happens. 

Most of the rime it works fine, but rarely it happens.

It was self fixed in active overview after 10 - 20 seconds. I think AAPS is loading data in that time.

closed 2022-03-07 by robertrub