nightscout/AndroidAPS #834: [3.0 beta2] Omnipod Eros. Sms delivery issue - Polish

betees82, 2021-11-02:
AndroidAPS 3.0-beta2
Remote: https://github.com/nightscout/AndroidAPS
Flavor: fullrelease
Nightscout version: 14.0.7
Engineering mode enabled

I'm not receiving bolus delivery reports if polish language is set. Problem disappears when language is switched to ENG.
I faced it also in 2.8 version.
[image]

MilosKozak, 2021-11-02:
it's a bug in libraries but i made a workaround in AAPS https://github.com/nightscout/AndroidAPS/commit/e715167254a63c00f6577e8af3a83409e7778b4b

closed 2021-11-02 by MilosKozak

betees82, 2021-11-04:
Thank you! I can confirm, 3b3 no issue with Polish

betees82, 2021-11-07:
Now after switching to Dash, some minor SMS Issue. 
[image]