nightscout/AndroidAPS #835: [DEV 2.8.2.9] eCarbs shown in AAPS, but wasn't entered by myself

Alamo04, 2021-11-03:
Only a small issue.
After I forgot my phone at home and comming back after a hour, there are eCarbs (1gr per tick) in my AAPS overview.
I never forgot my phone before, so I never saw that.

That seems to be ghost carbs, because they are only visible in overview, not in treatments, not in NS.
[AndroidAPS_LOG_1635943240758.log.zip](https://github.com/nightscout/AndroidAPS/files/7469233/AndroidAPS_LOG_1635943240758.log.zip)

Alamo04, 2021-11-03:
[image]

MilosKozak, 2021-11-04:
go to treatments - user actions
you will see all entered data

Alamo04, 2021-11-04:
Oh.. so I did a mistake at 3::23AM in the night 10g over 10h.
But if that was the reason, the shown data are also incorrect.

Thanks Milos, that was my day of mistakes.  😀

closed 2021-11-04 by Alamo04