nightscout/AndroidAPS #836: Progress pop-up windows for manual bolus immediately closed while bolus still progressing

vanelsberg, 2021-11-03:
**AndroidAPS 2.8.2.9-dev**
Build: aea868ac6 / Fullrelease
**Pump: DASH**

**Problem description:**
When starting a manual bolus (2u) the progress window popped, immediately showed a full bar, then closed.
The bollus command itself was executed without problems: seeing progress on wear, Pod ticking, DASH history confirming.

Repro:
This was a one-time occasion, could not repro.

**Additional Info**
This very much _looks like_ the problem AAPS had some month ago. The problems was caused by starting a manual bolus at the moment an SMB was running. When SMB finished it closed the progress window (for the manual bolus)?

MilosKozak, 2021-11-04:
this is close to impossible with current implementation

vanelsberg, 2021-11-05:
(*grin*) I'd take your word on that.
Problem is that I have seen it happen twice now. So maybe there is some other cause that closes the pop-up.
Will check the logs to see if they show anything that is "impossible" ;-)

MilosKozak, 2021-11-11:
i did some improvemnt recently but to be honest i'm not 100% it helps

vanelsberg, 2021-11-11:
Tnx. Have not seen it happen for a while as I described. So improvement may have helped.

One way to repro which may be similar in behavior:
Start a manual bolus. Then on the phone switch to active apps view and swipe/close AAPS. In will continue the bolus on the background (wear notifications confirm by showing progress). Opening AAPS does no longer show the progress pop-up (while bolus is progressing in the background).
Can reproduce this behavior without any problem.

MilosKozak, 2021-11-12:
when you close app (by swiping out) all activities are destroyed and there is no way ro restore it. it's android design
if you switch to another app and back bolus progress survive as activity is restored

closed 2021-11-12 by MilosKozak

vanelsberg, 2022-02-08:
@MilosKozak 
*Minor issues but it still is there?*
Do not think this should have high priority but still seeing this behavior now and then, also for current 3.0 release/dev branch.

Start a manual bolus from overview (so no closing of the app):
On starting the bolus progress opens with a flash of 100% yellow progress, then seams to start the bolus (grey 0% progress) immediately (within 1 secs) followed by closing the progress pop-up.
Bolus does proceed in the background without issues.

Just can not find a way to reproduce this. Have seen at least 2 users in Discord reporting the same.

CaptainBalou, 2022-02-08:
> @MilosKozak _Minor issues but it still is there?_ Do not think this should have high priority but still seeing this behavior now and then, also for current 3.0 release/dev branch.
>
> Start a manual [...]

I've had this as well with 3.0 betas. Setup here is DanaRS, Dexcom G6 with BYOD. But because I've found no way to reproduce it I couldn't create a log. Because the bolus in background was always delivered well I haven't put that much energy into it.

TranceCake, 2022-02-25:
I have seen this with both 2.8.x and 3.0 about 8+ times I think, of which 3 in 3.0