nightscout/AndroidAPS #839: [3.0-beta3] invalid NS profile (worked in beta2)

hartlijn, 2021-11-04:
right after installing beta3 (I came from beta2) below error popped up
[image]
.
[image]

It is referring to the BG target that has a value of 5.7 in nightscout.
As a result no profile is imported. I have quite some in NS, but importing stopped at the first profile.

MilosKozak, 2021-11-04:
pm me your ns address

hartlijn, 2021-11-04:
> pm me your ns address

I sent it via FB Messenger. regards, Joost

MilosKozak, 2021-11-04:
well it's a bug in NS profile (units are not explictly declared) but I made a workaround https://github.com/nightscout/AndroidAPS/commit/35c6ee38e5c8d55afd01be20e1999de939da4194

closed 2021-11-04 by MilosKozak