nightscout/AndroidAPS #842: Profile doesnt successfully activate if basal rate not adjusted

markvader, 2021-11-04:
[3.0-beta3] (previously on beta 1 & 2) I made a change to my profile, the only change i made was one of my ISF rates, then I activated it and completed a profile switch.

The profile shows as changed in the profile plugin and uploads to correct values to nightscout

However the profile is not updated in the profile bar on the home screen, nor is the change taken into account for calculations, and no profile change is shown on the main graph as it normally would.

I am using a Combo pump.

[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/7474603/AndroidAPS.log)

I have attached the log. At 09:47 I changed the ISF from 8.5mmol/U to as 9mmol/U (for the 7am to 10pm time period)

Line 38 shows the new rate

`09:47:16.997 [main] D/PROFILE: [LocalProfilePlugin.storeSettings():211]: Storing settings: {"defaultProfile":"LocalProfile","startDate":"2021-11-03T09:47:16.9960000Z","store":{"LocalProfile":{"dia":6,"carbratio":[{"time":"00:00","timeAsSeconds":0,"value":18},{"time":"11:00","timeAsSeconds":39600,"value":14.5},{"time":"16:00","timeAsSeconds":57600,"value":18}],"sens":[{"time":"00:00","timeAsSeconds":0,"value":8},{"time":"07:00","timeAsSeconds":25200,"value":9},{"time":"22:00","timeAsSeconds":79200,"value":8}],"basal":[{"time":"00:00","timeAsSeconds":0,"value":0.55},[other rates removed to reduce log],{"time":"23:00","timeAsSeconds":82800,"value":0.66}],"target_low":[{"time":"00:00","timeAsSeconds":0,"value":6},{"time":"08:00","timeAsSeconds":28800,"value":5.6}],"target_high":[{"time":"00:00","timeAsSeconds":0,"value":6},{"time":"08:00","timeAsSeconds":28800,"value":5.6}],"units":"mmol","timezone":"Europe\/London"}}}`

Line 135 and subsequent entries shows the old rate of 153 (i.e. 8.5mmol/U) being used:

`09:50:36.524 [RxCachedThreadScheduler-4] D/APS: [DetermineBasalAdapterSMBJS.invoke():73]: Profile: {"max_iob":7,"type":"current","max_daily_basal":0.72,"max_basal":2.16,"min_bg":100.8,"max_bg":100.8,"target_bg":100.8,"carb_ratio":18,"sens":153,"max_daily_safety_multiplier":3,"current_basal_safety_multiplier":4,"high_temptarget_raises_sensitivity":false,"low_temptarget_lowers_sensitivity":false,"sensitivity_raises_target":true,"resistance_lowers_target":false,"adv_target_adjustments":false,"exercise_mode":false,"half_basal_exercise_target":160,"maxCOB":120,"skip_neutral_temps":false,"remainingCarbsCap":90,"enableUAM":true,"A52_risk_enable":false,"SMBInterval":3,"enableSMB_with_COB":true,"enableSMB_with_temptarget":true,"allowSMB_with_high_temptarget":true,"enableSMB_always":false,"enableSMB_after_carbs":false,"maxSMBBasalMinutes":120,"maxUAMSMBBasalMinutes":70,"bolus_increment":0.1,"carbsReqThreshold":1,"current_basal":0.7,"temptargetSet":false,"autosens_max":1.2,"out_units":"mmol\/L"}`

The profile switch doesn't successfully change, if adjustments are only made to IC, ISF or Targets.  It only successfully updates if I change a rate in the basal profile.

MilosKozak, 2021-11-04:
* pump connections was in bad state
```
09:47:23.725 [Thread-6675] W/info.nightscout.androidaps.plugins.pump.combo.ruffyscripter.RuffyScripter: Disconnect not happy
java.lang.NullPointerException: Attempt to invoke virtual method 'void org.monkey.d.ruffy.ruffy.driver.PumpData.incrementNonceTx()' on a null object reference
	at android.os.Parcel.createException(Parcel.java:2077) ~[na:0.0]
	at android.os.Parcel.readException(Parcel.java:2039) ~[na:0.0]
	at android.os.Parcel.readException(Parcel.java:1987) ~[na:0.0]
	at org.monkey.d.ruffy.ruffy.driver.IRuffyService$Stub$Proxy.doRTDisconnect(IRuffyService.java:216) ~[na:0.0]
	at info.nightscout.androidaps.plugins.pump.combo.ruffyscripter.RuffyScripter.disconnect(RuffyScripter.java:199) ~[na:0.0]
	at info.nightscout.androidaps.plugins.pump.combo.ComboPlugin.disconnect(ComboPlugin.java:252) ~[na:0.0]
	at info.nightscout.androidaps.queue.QueueThread.run(QueueThread.kt:133) ~[na:0.0]
```

* no other pump logging is in log file

markvader, 2021-11-04:
okay, have added the additional pump logging back in.

At 12.57 I changed IC from 14.6 to 14.5
[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/7474961/AndroidAPS.log)

`12:57:27.952 [main] D/PROFILE: [LocalProfilePlugin.storeSettings():211]: Storing settings: {"defaultProfile":"LocalProfile1","startDate":"2021-11-04T12:57:27.9510000Z","store":{"LocalProfile1":{"dia":6,"carbratio":[{"time":"00:00","timeAsSeconds":0,"value":18},{"time":"11:00","timeAsSeconds":39600,"value":14.5},{"time":"16:00","timeAsSeconds":57600,"value":18}],"......`

MilosKozak, 2021-11-04:
fixed by https://github.com/nightscout/AndroidAPS/commit/ebeac72e7412f1ec9e372962d0ce96f0ce64dee7

closed 2021-11-04 by MilosKozak