nightscout/AndroidAPS #844: Overview should display BG data once it's configured

rustymonkey, 2021-11-04:
When the BG data source is set up, the BG data should appear on the Overview page--even if other aspects of the loop aren't configured. Otherwise, users are given the impression that there's a problem with the configuration/functionality of their BG source. It's a pretty minor issue since it resolves itself once the rest of the loop is configured, but it's nonetheless confusing.

Replicated by:
1 Importing 2.8 profile to a clean install
2 Changing BG source to NS OR Xdrip
--> BG data appears in the nightscout or Xdrip tabs (if they're enabled)
--> BG data doesn't appear on the 'Overview' screen

Tested on: 3.0. Beta 3
Originally raised at: #797

MilosKozak, 2021-11-04:
there must be an active profile. without profile it's not possible to calculate many things rendered on overview. There should be clear message "NO PROFILE" visible
As update to 3.0 changes DB there is no active profile migrated and must be done again

closed 2021-11-04 by MilosKozak

rustymonkey, 2021-11-04:
Why not render BG data but not any of the profile-dependent data?

MilosKozak, 2021-11-04:
well why yes? AAPS is not an CGM monitor

rustymonkey, 2021-11-08:
Even though it's not a CGM, it is a 'dashboard' for AAPS, and the manner it which it displays sets an expectation that BG data should appear if it's been configured. As an analogy, if a car has overheated and can't drive, the dashboard still indicates the status of the gas tank, and Android Auto will continue to display.

Again, it's minor and users will eventually figure this out--it's just an extra bit of friction.