nightscout/AndroidAPS #846: [3.0-beta3] Small fixes on the Treatments page

fjpezuela, 2021-11-04:
When entering the treatment page, some links are not displayed correctly in vertical view, as the text is too long (Spanish language). The only way to see it correctly is to change the position of the screen to horizontal.
Version: APPS 3.0-beta3

[image]

Sample of the problem

[image]

When you are in any of the options marked in the red square and you turn the mobile horizontally, it automatically goes to the first option marked with an arrow. The option in which we are positioned is not maintained.

[image]

Thanks a lot for your support

MilosKozak, 2021-11-04:
the problem is too long translation of IOB

fjpezuela, 2021-11-04:
If that is. Is it possible that it shows in two lines like you do in the other options? Another option is to try to shorten the translation.

MilosKozak, 2021-11-04:
what language is it?

fjpezuela, 2021-11-04:
Spanish (Spain)

MilosKozak, 2021-11-04:
i updated the translation

closed 2021-11-04 by MilosKozak