nightscout/AndroidAPS #847: Unable to cancel alarm during Dana-i bolus delivery

MichaelJohnSmith, 2021-11-04:
I use a 60s/U bolus rate, so a 10U bolus takes about 10 minutes.  This morning I replaced my Libre2 sensor and was waiting the 60 min sensor statup.  After 30min, AAPS alarms due to missed BG readings.  Normally this is not a problem, as the alarm can be snoozed.  This time, my 10 min bolus was in progress and I could not access the snooze button.  I was able to turn down the sound to hide the alarm but was unable to use the phone to take calls or play music/podcasts until the bolus completed.

Is there any way to access high priority functions such as alarm snooze while the bolus dialog has frozen the UI?

Version 3.0-beta3 (but I suspect same in all versions).
Dana-i pump
Libre2 sensor via patched app and Xdrip+
Samsung A20e with Android 11

[image]

MilosKozak, 2021-11-04:
it's not possible at the moment. this request go against current UI design: block UI during bolus.
Marking as enhancement but there are no plans to change it right now

AdrianLxM, 2021-11-06:
A possible workaround: swipe away the system notification to stop the alarm.

closed 2026-10-15 by MilosKozak