nightscout/AndroidAPS #852: DEV/BETA URGENT !!! Don't stop closed loop  !!!

robertrub, 2021-11-06:
@MilosKozak Please DO NOT stop closed loop in tje middle of the night ! I'm using dev (maybe 1 or 1.5 days old) and suddenly engineering mode stoped (probably because there is a new dev version) in the middle of the night and I woke up with a hypo alert because it. You want me to build a dev at 1AM ???
Please send an alert first, sth like "New version available, closed loop will stop in 24 hours !" and stop loop in the middle of the DAY and not night.

MilosKozak, 2021-11-06:
what version and what hapen to engineering mode?

robertrub, 2021-11-06:
Sorry, I don't have the build number _screen captured_. I'd built it probably 1-2 days ago at most. I can search Dropbox waste basket to get the date time stamp of the file if you need.

The engineering file gets deleted. This happened severel times but I supposed it was related to a program update. This night/morning, it had disappeared (but no program update at 1AM of course). I woke up with a 63 going down and xDrip alert. AAPS was on open loop (as there was no more an engineering file) and was proposing to zero temp. Don't have precise data as I was half asleep.

Alamo04, 2021-11-06:
@robertrub 
I never get that with all of my tested dev versions.

Don't think that this is an AAPS Issue, think you have a problem with your phone.
Check your phone settings, is a secure program working and deleted the engineering mode file ?

robertrub, 2021-11-06:
@Alamo04  Interesting suggestion, tnx. I don't have an active "cleaning" app (CCleaner is installed but stopped) but maybe Android is doing some "maintenance" and deleting 0 bit files...
I added some text in the engineering file so that it is not empty. I'll see if it stays stuck or still goes away...

RalfCGM, 2021-11-06:
I had closed loop stopped on the 4th of november. I had a couple of messages that dev was going to expire the 14th of nov, but it happened the 4th. My Engineering file wasn't deleted. So maybe a different issue?

Alamo04, 2021-11-06:
Different issue, because here was the file deleted and without file,  the dev has to stop working.
That is intend.

MilosKozak, 2021-11-06:
AAPS doesn't delete this file itself

robertrub, 2021-11-06:
Tnx. Not an AAPS problem then. Probably Android "cleaning" empty files.

Closing this issue.

closed 2021-11-06 by robertrub