nightscout/AndroidAPS #853: 3.0 beta3

RalfCGM, 2021-11-06:
Weird issue with the homescreen. 

BG data is coming in. Calculations are being made, as is visible on the message / announcements screen, but on the aaps homescreen the data isn't updatong, and time looks like running back. (amount of minutes of last bg reading is going up faster than the actual time). 

See screenshot and the timestamp of the screenshots.

Restarting aaps and clearing cache doesn't help. 

After 10 more minutes the issue disappeared. (last screenshot). 
[image]

[image]
[image]
[image]

Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/EN/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

RalfCGM, 2021-11-06:
[AndroidAPS._2021-11-06_03-12-13_.1.zip](https://github.com/nightscout/AndroidAPS/files/7489636/AndroidAPS._2021-11-06_03-12-13_.1.zip)
[AndroidAPS._2021-11-06_03-12-13_.0.zip](https://github.com/nightscout/AndroidAPS/files/7489637/AndroidAPS._2021-11-06_03-12-13_.0.zip)

mick711, 2021-11-06:
Not à New issue i sera two ôther issue like this and i have this issue yesterday with thé last dev. Il seem that nobody would like to explain or solve this problem. If needed i have thé log files

AdrianLxM, 2021-11-06:
@mick711 would you have a link to your GitHub issue please?

MilosKozak, 2021-11-06:
turn on event logging and update to 3.0-b4 (or latest dev)
then provide logs again if the issue reappear

RalfCGM, 2021-11-06:
I'll add that this issue seems to appear after missed readings.

Ill update to latest dev and see.

RalfCGM, 2021-11-09:
Happening again on beta5. 
[image]
[image]
[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/7507164/AndroidAPS.log)

MilosKozak, 2021-11-10:
i revorked UI updating un dev 
will be in beta6 or current dev

MilosKozak, 2021-11-20:
@RalfCGM how about beta6?

RalfCGM, 2021-11-21:
Beta 6 doesn't work ok for me. See issue https://github.com/nightscout/AndroidAPS/issues/952

used beta 6 a short time (2 days) but reverted back tot master.

The time i used it the issue didn't appear.

closed 2021-11-29 by MilosKozak