nightscout/AndroidAPS #854: DASH : Not urgent / Text correction while injection

robertrub, 2021-11-06:
When Omnipod (EROS and DASH) is injecting, we don't know the amount or percentage of the dose already injected. We just have the progress bar.

The injection window shown has 2 lines of text and the 2nd one is not correct. It **said** (EROS) (Milos changed it) and **says** (DASH) (needs to be changed) the _total_ amount has been injected. Information is not exact.

It would be great if, on EROS and DASH, we could have the real amount already injected (but that info seems difficult to have). The least would be to change the text in DASH and put the same one as EROS. It won't be informative but at least, it wont be false.

[image]

x U will be injected

[image]

x U has been injected.

avereha, 2021-11-08:
Do you still see this issue for Dash?
What version?
Do you see the same issue also for English?

robertrub, 2021-11-09:
I can only see in French and it still says The units have been injected (while the injection is still going on). 

[image]

[image]

avereha, 2021-11-09:
Yes, the text in English is correct, the translation in French is not. 
I think the fix should be implemented in Crowdin.

robertrub, 2021-11-09:
Ok, will go and correct it in Crowdin. Pity thst we can't have the real amount injected... ;)
Closing this issue.

StephenBrown2, 2021-11-10:
I sympathize with you, I would like the secondary text to update with the current dose progress as well.

robertrub, 2021-11-11:
Hi @avereha

I'm reopening this issue as the text is NOT in Crowdin. It must be in rhe code directly (just a supposition). Czn you vetify please ?

Request bis : As the number of tics/sec (and 0.05U/tic) is constant for DASH and EROS , why can't we have the real value injected thete (like the other pumps ?)

avereha, 2021-11-12:
It's here: https://crowdin.com/translate/androidaps/1072/en-fr#49728

The reason Dash does not check delivery status while in progress is to preserve battery life on the pod.
The delivery time also depends on the basal rate at the time(if the basal rate is higher, delivery time will be longer).

robertrub, 2021-11-12:
@avereha Tnx for the link. I don't know how I missed it. The text is corrected, waiting the validation of @Philoul 
The shown "injected units" doesn't have to be that precise. A simple x U = y seconds. Showing a % or U just by calculation (without any pod participation) will be better than a fix number and very unprecise progression barre.

robertrub, 2021-11-12:
Coming back to the subject. The link that you showed is in Omnipod commun. I'm almost sure that it points to the title **of the window** and NOT to the text inside rhe window.  In commun, EROS and DASH should show the same text but they do not. It is either in the code or not translatable (not visible in Crowdin)

Philoul, 2021-11-12:
> @avereha Tnx for the link. I don't know how I missed it. The text is corrected, waiting the validation of @Philoul

This string (bolusdelivering) is used by other pumps (for ex Dana, combo...) to show the updated amount of insulin during injection.
I think we cannot update this string to use it in another context (global amount during the whole injection)...

closed 2021-11-14 by MilosKozak