nightscout/AndroidAPS #855: AAPS 2.8.2.9-dev, Yellow Warning Triangle in Main Screen

Thomas-Wolff, 2021-11-06:
Since the update to AAPS 2.8.2.9-dev build 2b021ffaf I have a yellow warning triangle in the top left corner of the main screen and I have absolutely no information what this is supposed to signal.  AAPS runs without any problem.
I am not sure how far this is an issue, but I would appreciate any information about this.

[image]

[image]

Best regards Thomas

robertrub, 2021-11-06:
I didn't have it in the last build

[image]

Then I turned _ON_ backfill BGs from NS (I had only half the screen of the BG curve) and then I got the red triangle. I'm not sure if they are related. Un checking the _NS BG backfill_ did not make the triangle disappear.

[image]

MilosKozak, 2021-11-06:
https://www.facebook.com/groups/AndroidAPSUsers/posts/3115136028707793/?comment_id=3130352127186183&__cft__[0]=AZVxHcA30glk-Zd30GDgB95Ot_fZFVzhxb501quwrlOCHvdpDpA0PPEomprW8fbyrTOK7i1evSkbQM7NeYYLwiJSriiJOIKSNbuRbSM50b7G6cxynvt65fDEKtw2FO15VWjjwDCeoRZPdXjYdLmdj2IKWFcfNMvOXjXw4ToeGc6wbXGhipIIyN5TpMiRILzcwY0&__tn__=R]-R

closed 2021-11-06 by MilosKozak

robertrub, 2021-11-06:
No FB for me (it is not possible to select _Do not use tracking cookies_ thus I refuse to use FB). What does the link says ?

Alamo04, 2021-11-06:
That the triangle is intend.
It shows the quality of your bg data, if it goes to red closed loop will stopped.

Milos Syd there:
> there is new icon in 3.0-beta to report BG data quality. it looks like this. Yellow means data must be recalculated to 5 min interval, red means there are BG duplicates preventing AAPS from running in closed loop mode. If you are getting red one, see: https://github.com/nightscout/AndroidAPS/issues/689
https://github.com/nightscout/AndroidAPS/issues/772