nightscout/AndroidAPS #857: Loop Diasbled due to constraints

vanelsberg, 2021-11-06:
**AndroidAPS 2.8.2.9-dev**
Build: 2b021ffa
Flavor: Fullrelease

**Problem description:**
Getting RED warning icon in overview
Loop displays on the "Last run" field: LOOP DISABLED BY CONSTRAIMTS BgQualityCheck: Doubled values in BGSource

Per configuration, AAPS should ignore BG values from NS. _Only BG sources is xDrip as per configuration_

**Additional information**

- AAPS/NSClient is set to "Upload data to NS" (all other options switched "off")
- AAPS BG Source is set to xDrip+ (release version 2021.03.30, G6-client).
- xDrip+ is synchronizing with Nightscout, also sending BG values.
- No other sources are uploading to Nightscout

This configuration has been working for a long time.
_As BG sources for AAPS is set to "xDrip" I expect it to ignore BG values from NS (nor should it download anything else)_

But with the latest QulityCheckPlugin update (commit#2b021ffafc0c4f01df65b7dfe28c3d0e51cc088f) something changed making AAPS to decide disablling loop due to _BG values from NS_?

**Logfile:**
[AndroidAPS_LOG_1636205974971.log.zip](https://github.com/nightscout/AndroidAPS/files/7490608/AndroidAPS_LOG_1636205974971.log.zip)

**Screen-prints:**
[image]

[image]

vanelsberg, 2021-11-06:
Please also see my comments on
https://github.com/nightscout/AndroidAPS/issues/689?

robertrub, 2021-11-06:
And the red triangle stays there even when _Backfill BGs from NS_ has been turned off. Closing AAPS, clearing cache memory doesn't get rid of the red triangle.

Alamo04, 2021-11-06:
If that is an issue with latest dev, then the new BG Quality Check is possible the reason ?
https://github.com/nightscout/AndroidAPS/commit/2b021ffafc0c4f01df65b7dfe28c3d0e51cc088f

RalfCGM, 2021-11-06:
Ive got the same issue. Loops isn't able to run. Data coming from xdrip. Bg Data table in xdrip and, bg data table in aaps doesn't show doubles.

MilosKozak, 2021-11-06:
i made the triangel clickable in dev branch

MilosKozak, 2021-11-07:
you may try update to latest dev and clear db

vanelsberg, 2021-11-07:
@MilosKozak @RalfCGM I think the problem is cause by executing receive/backfill CGM data.
Both on AAPS and/or xDrip.

closed 2021-12-07 by MilosKozak