nightscout/AndroidAPS #858: Dev DASH/ Pod inactivated by itself

robertrub, 2021-11-06:
Last dev version, using DASH pod

[image]

I get a sound alert from AAPS, _pod is unreachable_ (phone was charging at 5m from DASH) at around 14:31. I mute the alert and wait for the pod to connect but nothing happens. I check and DASH has been suspended all by itself...

[image]

Here is the copy of the alert

[image]

And below ate the logs.

avereha, 2021-11-07:
Do you have automation that updated the basal rates?

Updating the basal rates for Dash involves: suspend delivery, set a new basal rate.
In this case, the communication failed after the first step.

robertrub, 2021-11-07:
Hi Andrei 

Yes, I have automation that change profile % if high delta or high bg.

The strange part is that, when phone and pod were near again, the connection didn't kick in and refreshing pod status did nothing either (if my memory is good...)

I had to "unsuspend" pod and then be zble to refresh.

robertrub, 2021-11-09:
Sorry for the delay. I'm very late on everything at the moment...

I got the alert ringing on AAPS (Mute, wait 5 min etc). I muted it and tried to actualize the pump directly after that. 

The 15 minutes wait is not logical as the alarm that "pulls" the user to AAPS and s/he needs to silence it. I don't know why it didn't actualize. Is actualizing not possible if the pump is suspended ?

avereha, 2021-11-25:
I think this can be closed after https://github.com/nightscout/AndroidAPS/pull/984

closed 2022-03-07 by robertrub