nightscout/AndroidAPS #860: NSClient shows TBR in percents only.

shafiro, 2021-11-06:
NSClient 2.8.2.9 dev
Build: f49c93f-2021.11.4-12:37
Remote: https://github.com/nightscout/AndroidAPS
Flavor: nsclientdebug
Nightscout version: 14.2.4
Engineering mode enabled
Device: Xiaomi MI 9, Android 11, MIUI 12.5.1

NSClient shows TBR in percent.
BR is shown in U/H.
AAPS shows both TBR & BR in U/H. 
[image]

shafiro, 2021-11-06:
Similar behavior described in docs under
"AAPSv2 watchface - Legend", but not for NSClient. 

"G - basal rate (shown in U/h during standard rate and in % during TBR)"
https://androidaps.readthedocs.io/ko/stable/CROWDIN/ko/Configuration/Watchfaces.html#aapsv2-watchface-legend

shafiro, 2021-11-06:
I have tested this with NSClient 2.8.2, which always showed TBR with U/H, and it also moved to "%".
So change is coming from AAPS or NS but not a problem with NSClient.

MilosKozak, 2021-11-07:
fixed by https://github.com/nightscout/AndroidAPS/commit/e14fc44957e2339d2c63726d548f144b0dd766e2

closed 2021-11-07 by MilosKozak