nightscout/AndroidAPS #861: NSClient [3.0 beta4] No Dash driver- can't change TBP

betees82, 2021-11-06:
NSclient 3.0-beta4
Remote: https://github.com/nightscout/AndroidAPS
Build: 9202b91c5-2021.10.30-12:09
Flavor: Nsclient
Nightscout version: 14.0.7

Can't change TBR via NSclient
Massage: Data from another pump. Change pump driver, to reset pump state.

[image]

MilosKozak, 2021-11-06:
you cannot set TBR from NSClient. with any version

betees82, 2021-11-06:
No it looks like I'm not able to change TBR directly in AAPS. 
[image]

betees82, 2021-11-06:
[https://photos.app.goo.gl/hRDHGwdF3yCby6dj8](url)

betees82, 2021-11-07:
TBR is not synchronizing to NSClient

[image]

betees82, 2021-11-07:
Looks like issue is with T:0,00 U/h
[image]

MilosKozak, 2021-11-07:
i improved TBR sync in https://github.com/nightscout/AndroidAPS/commit/e14fc44957e2339d2c63726d548f144b0dd766e2

closed 2021-11-07 by MilosKozak