nightscout/AndroidAPS #862: Dana I not showing TDD

pbloom209, 2021-11-07:
Reporting bugs
--------------
Using Beta3 ver 4 and previously ver3
Used these versions from the time they have each been released.
Using Dana I
Same pump all the way through.

I find that it does not total the units for TDD.

https://user-images.githubusercontent.com/37948551/140643805-061f8f65-38e6-47a1-88d6-16b27652a0b1.mov

Philoul, 2021-11-07:
How many days do you have in aaps database? 
TDD is not calculated with pump informations, but with data recorded in aaps database...

MilosKozak, 2021-11-07:
fixed by https://github.com/nightscout/AndroidAPS/commit/3bbf50ca3e3193c99f860885b29db57e4d2fffd6

pbloom209, 2021-11-07:
I did the code change and reinstalled.

I ran the TDD after a few hours.  See attached.
Seems to be the same issue

https://user-images.githubusercontent.com/37948551/140664175-c43919c3-cff4-4619-b5a1-3a1ed28484eb.mp4

.

MilosKozak, 2021-11-07:
try it after one day

mxsrm, 2021-12-06:
@MilosKozak 
The issue is not fixed (3.0-beta9): after one day it will only show the TDD for the day that has passed, not the rest:

[image]

MilosKozak, 2021-12-07:
i checked it again and it works with Dana-i
wait at least 2 days after you changed pump

mxsrm, 2021-12-08:
@MilosKozak After 3 days it still does not show more than the days that AAPS has data for anyway; nothing that is loaded from the pump is used:

https://user-images.githubusercontent.com/7951225/145269035-3295e816-473e-4276-b43d-d37fa29bbf55.mp4

mxsrm, 2021-12-13:
Yes, the pump shows normal data as far back as I want (checked till start of november). After 6 days it still only shows data that AAPS collected itself.

[image]

mxsrm, 2021-12-13:
Pump history browser shows the data without problem, it's just never used in TDD tab:

[image]

MilosKozak, 2021-12-13:
ok. reload history from AAPS and provide logs. we'll see

MilosKozak, 2021-12-13:
be sure you are using latest version

mxsrm, 2021-12-13:
> be sure you are using latest version

Beta or dev?

mxsrm, 2021-12-13:
Log using beta9:

[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/7705586/AndroidAPS.log)

MilosKozak, 2021-12-13:
i see loaded data 5/12 - 12/12

mxsrm, 2021-12-14:
For all older entries it just notes

```
18:04:56.867 [main] D/CORE: [TDDStatsActivity.onResume$lambda-18():209]: EventDanaRSyncStatus: 05/12/2021 00:00 dailyinsulin
18:04:56.915 [Binder:15993_3] D/PUMPBTCOMM: [BLEComm.processMessage():859]: <<<<< REVIEW__DAILY B2 12 03 15  0C 04 03 B6  14 50 18 06
18:04:56.918 [Binder:15993_3] D/PUMPCOMM: [DanaRSPacketHistory.handleMessage():137]: History packet: 3 Date: 04/12/2021 00:00 Code: 80 Value: 6150
18:04:56.925 [Binder:15993_3] E/PUMP: [PumpSyncImplementation.confirmActivePump():83]: Ignoring pump history record  Allowed: 04/12/2021 14:36:34 DanaRS TKK00197FB Received: 1638572400000 04/12/2021 00:00:00 DanaRS TKK00197FB
```

"Ignoring pump history record  Allowed: 04/12/2021 14:36:34" so it discards all data before the new install of AndroidAPS started?

MilosKozak, 2021-12-14:
only data from settin up pump in AAPS is accepted
do you see these 4 days?

mxsrm, 2021-12-14:
Yes the days since setting up AAPS are displayed correctly. I wasn't aware, that the days before are not taken into account. Then it works as intended.

closed 2021-12-14 by MilosKozak