nightscout/AndroidAPS #863: [3.0-beta4] Problem with COBs when canceling a bolus from the wizard

fjpezuela, 2021-11-07:
AAPS 3.0-beta4.

I have used the bowling assistant to put in a bolus of food. After accepting the calculations, before the connection to the pump was established I canceled it. Then I used the wizard again to do the calculation again and I let it bolus.

The canceled bolus insulin has not been added, but the carbohydrates that I put in the wizard were added, so they have been counted twice. I have had to cancel the carb entry manually.

I leave some screenshots where you can see what has happened:

[image]

[image]

The meal had 70g of carbohydrates in total, and 110g have been counted in AAPS

Regards, Fran.

Philoul, 2021-11-07:
I don't see anything wrong in your screenshot...
The validation of calculator "validate" Carbs and bolus, then the cancel only cancelled the bolus before injection, You did not cancel carbs.
The only strange this is the second time you entered Carbs in bolus calculator, the bolus should have been more important (about 7.3U  for 80g and not 3.6U for 40g), probably because you entered too quickly in Calculator (not enough time to update cob value of 1rst entry to integrate it in calculation...

MilosKozak, 2021-11-07:
i changed the behavior to store carbs only if bolus was successfull
https://github.com/nightscout/AndroidAPS/commit/dcf3c9cc57b0a2b63af9ee418dafee82fefef95b

closed 2021-11-07 by MilosKozak