nightscout/AndroidAPS #864: 2.8.2.9-dev: Profiles in NS deleted when going through setup wizard

jumalakarhu, 2021-11-07:
Clean install of AAPS to a new phone without importing any settings. Using the wizard and connecting to NS results in deletion of all existing profiles in NS. This is because AAPS by default is set to upload data and to receive/sync profiles. Since iOS Loop cannot import setting and are most likely to  use virtual pump in the beginning this should either be declared in wizard and/or in documentation. Another solution will be to not sync profiles as default.
Tested with most recent version 2.8.2.9-dev and also previous dev (6?) version.
Can provide logs if necessary, but will be messy since I did not remove old files from before testing again today.

MilosKozak, 2021-11-07:
It's not true.
* if you go through SetupWizard, enter NS credentials (see NS connected below), and continue to Profile, you'll start with profiles from NS
* if you don't connect to NS and continue to Profile, then new empty profile is created and sent to NS

even in second case you still can go to NS profile editor, find previous profiles and do "Clone". This way you make old profiles "newer" and they will be sent to AAPS

closed 2021-11-07 by MilosKozak

jumalakarhu, 2021-11-07:
I'm not trying to do anything else than testing rigorously and trying to anticipate what questions new users will ask. Most (all) testers are already aaps users and go directly to import settings.
I did exactly what you said. New profile is sent, but all existing profiles in NS were deleted. If I could "manage" this others can too.
Still I could get the old profiles in NS - thanks.

MilosKozak, 2021-11-07:
try it again.
I did a test before my reply to check if it's right
Importing settings will be more tricky because you import data with old timestamp. in this case probably manual intervention will be needed anyway

MilosKozak, 2021-11-07:
if you don't use NS for editing profiles you can disable accepting profiles from NS. then local profile always stay intact