nightscout/AndroidAPS #867: [3.0-beta4] TT targets display the wrong value or units on Gear Tab

MichaelJohnSmith, 2021-11-07:
TT created from the Target button show the correct value but the wrong units
TT from automation shows the wrong value but the correct units

[image]

[image]

MilosKozak, 2021-11-08:
fixed by https://github.com/nightscout/AndroidAPS/commit/225797f430aa1324190a47de42d4957eda4b8364

closed 2021-11-08 by MilosKozak

MichaelJohnSmith, 2021-11-08:
I got my errors swapped over above.
TT from target button now OK

However, in 3.0-beta5  TT from automation still shows the correct value but the wrong units

MilosKozak, 2021-11-08:
thanks! https://github.com/nightscout/AndroidAPS/commit/c684f12aa7a42decaa260231b33e5b9e6dc95c3e