nightscout/AndroidAPS #874: [last dev] Carbs from wizard not recorded in database

Philoul, 2021-11-08:
I don't know if it's a side effect of https://github.com/nightscout/AndroidAPS/commit/dcf3c9cc57b0a2b63af9ee418dafee82fefef95b but this morning my carbs entered in CALCULATOR was not recorded in database (see screenshot and log) (First time I see missing carbs in database)
[AndroidAPS._2021-11-08_08-41-23_.0.zip](https://github.com/nightscout/AndroidAPS/files/7495715/AndroidAPS._2021-11-08_08-41-23_.0.zip)

[image]

MilosKozak, 2021-11-08:
there was a crash at 08:41:25.913

Philoul, 2021-11-08:
Very strange in log file the line before crash is:
```
08:41:25.862 [main] D/PUMP: [OmnipodDashPodStateManagerImpl.load():665]: Creating new Pod state
```
Could it be the reason of the crash ? (How can I have this line (`Creating new Pod state` from Dash driver with an Insight driver selected...)

Philoul, 2021-11-08:
Got a crash when I tried to manually enter Carbs from Carbs button with the same strange OmnipodDash line before

Alamo04, 2021-11-08:
Confirmed, with dev from 00:12 AM
If I try to enter carbs, AAPS crashed, restarted and Carbs are not entered. Tryed again with same result.

robertrub, 2021-11-08:
I confirm the missing carbs too @MilosKozak.
Latest dev built this morning

[image]
 
At noon I use the wizard to calculate and inject my bolus. The insulin and the calc line are in treatments but not the carbs. I saw that something was wrong when my 2nd graph showed onlt IOB and no COB. I didn't have a crash.

[image]

[image]

MilosKozak, 2021-11-08:
fixed by https://github.com/nightscout/AndroidAPS/commit/2a2bb4edc63df8fdb3cd05a17b1967c56babd604

closed 2021-11-08 by MilosKozak

naromero77, 2023-06-09:
I had something like this happen today except it was with DASH pod and the 3.2.0-beta1. I will try to get logs and screen snapshots next time. I have been testing AAPS on and off for over a month for my son. First time such a thing happened.