nightscout/AndroidAPS #881: [AAPS3.0_beta5] NSClient - No Plot on Main Screen

old-square-eyes, 2021-11-09:
No plot after full sync with NS, no other metrics.

Creating new issue since quite a few things have been fixed/changed from the initial issue.

[image]

Data exists in history

[image]

old-square-eyes, 2021-11-09:
Old issue: https://github.com/nightscout/AndroidAPS/issues/808

MilosKozak, 2021-11-09:
NO active profile is the issue

old-square-eyes, 2021-11-09:
@MilosKozak - I cannot set a profile. Please see #808, and your comment on https://github.com/nightscout/AndroidAPS/issues/807#issuecomment-962518009 - I believe you told me I cannot have an active profile. I have tried everything possible to set a local profile. It is not possible.

old-square-eyes, 2021-11-09:
I think I must be missing something #808 even has a video posted of me trying to set a profile.

MilosKozak, 2021-11-09:
we were already talking about it, didn't we? you must use full AAPS or PUMPCONTROL to create effective profile switch

old-square-eyes, 2021-11-09:
@MilosKozak - so are you saying I cannot have a plot in NSclient, by itself?

old-square-eyes, 2021-11-09:
I understood NSclient to be a standalone client... to visualise NS data and use care portal. Why do I need Pump control or AAPS to see the plot.

MilosKozak, 2021-11-09:
bucause no other tool supports new features at the moment

closed 2021-11-09 by MilosKozak

old-square-eyes, 2021-11-09:
@MilosKozak there seems to be a categorical misunderstanding here. Are you saying NSclient is no longer functional standalone?

old-square-eyes, 2021-11-09:
If so, why am I spending hours testing it?