nightscout/AndroidAPS #888: [3.0-beta5] - SMS messages coming twi

streethawk22, 2021-11-09:
Reporting bugs
--------------
- **Note the precise time the problem occurred** and describe the circumstances and steps that caused
  the problem
- Note the Build version (found in the About dialog in the app, when pressing the three dots in the
  upper-right corner).
- Obtain the app's log files, which can be found on the phone in
  _/storage/emulated/0/Android/data/info.nightscout.androidaps/_
  See https://androidaps.readthedocs.io/en/latest/EN/Usage/Accessing-logfiles.html
- Open an issue at https://github.com/nightscout/AndroidAPS/issues/new

streethawk22, 2021-11-09:
SMS messages coming always two times, but only to the second number in the SMS Commands setting list.

MilosKozak, 2021-11-11:
I cannot replicate this issue