nightscout/AndroidAPS #889: Pump/Rileylink connection error messages persist after the failed connection has been resolved

rustymonkey, 2021-11-09:
If Pump or Rileylink connectivity is lost and subsequently regained, errors/notifications persist regarding the self-resolved problem. e.g.

1) Red notification bar that appears in some cases at the top of the dialog with a [snooze] button
2) Failed basal adjustment dialogs with the [Mute] button

The situation re. 2) is even worse since if the user ignores the warnings e.g. for 20 minutes, then upon entering AAPS they must click the 'OK' button 4x in order to make the dialog disappear (I presume that it appears every 5 minutes and that after 20minutes of failed basal updates, there are 4 dialogs superimposed upon one another).

Tested on AAPS 3.0 B5
Related to #884

rustymonkey, 2021-11-15:
Related to #932 (from a design perspective--another case of error messages inconsistent with the actual status).

rustymonkey, 2021-11-15:
Another side effect of the current design is 'Alarm fatigue' e.g. the Failed basal adjustment dialog / alarm appears upon a failed update. After 30 seconds, the user acknowledges the alarm and checks AAPS > Medtronic Tab to see what the problem is, but there is no longer any problem--it shows that the RL and Pump connected, and within a minute the basal adjusts on its own. i.e. the system appears to generate an alarm right away even though the connection issue often self-corrects within a minute. This results in alarm fatigue and the user either starts ignoring alarms or sets volume to 0 because of the frequency of irrelevant alarms. Perhaps such alarms should only trigger if either:
- the failure persists even after a retry x minutes later fails
- the failure will results in out-of-range BG within the next y minutes

MilosKozak, 2026-10-11:
This is fixed in the current dev version. A failed basal profile update now shows one notification instead of a stack of alarm screens, and it is removed by itself after the next successful update. "Pump unreachable" is also cleared when the pump answers again. Please open a new issue if you still see old errors with a current version.

closed 2026-10-11 by MilosKozak