nightscout/AndroidAPS #890: [3.0 beta5] Issue with TBR value -Dash

betees82, 2021-11-09:
AAPS 3.0-beta5
Remote: https://github.com/nightscout/AndroidAPS
Build: 917de7261-2021.11.09-18:49
Flavor: AAPS
Nightscout version: 14.0.7
Omnipod Dash

When there is no overwritten TBR, and basal rate is according to profil. In Overview screen of AAPS it is shown as TBR 0U/H. When it is such situation, it is not possible to use CANCEL TBR in ACTIONS tab.
In Dash Tab, there is no value in TBR, In NS basal rate is visible as normal rate, acording profile setings

[image]

When TBR is different it is working ok, and it can be cancelled, but cancelation provides to TBR 0 U/H, not possible to set own TBR value.

[AndroidAPS (3).log](https://github.com/nightscout/AndroidAPS/files/7508064/AndroidAPS.3.log)

avereha, 2021-11-09:
I will need logs from the last >3 days. 
Did you switch between Dash and Eros(or other pumps)  after activating a pod?

betees82, 2021-11-09:
Ok i'll attach long log, but now pod is just activated today so i need to wait few more days.
Yes, we use both Eros and Dash. Currently we're on Dash.
We are now on 2nd dash. And this issue we faced when we started DASH. so 3 beta 4. Now updated to beta 5 and same issue.
Let me know direction what is needed? Best would be whloe POD (3 days) history? I'll also send more screenshots.

betees82, 2021-11-09:
Here is situation with vale different than 0 U/h and different that set in profile.

[image]

avereha, 2021-11-09:
This buggy TBR was set sometime in the last 3 days. If you still have them, please attach the logs for the last 3/4 days(since the 6th of November).

When did you start with Dash? After activating a Dash pod, did you switch back to Eros?

The only solution that I know for this problem to go away is to wait 3 days(drop the pod and wait). Changing the AAPS version will not help.

I'm also interested in the pod history screenshots if you can share them. If there are errors, please scroll down to the first line that doesn't have an error. If all lines have errors, please share the first line in history.

avereha, 2021-11-09:
It's also possible that this issue is a duplicate of https://github.com/nightscout/AndroidAPS/issues/861

betees82, 2021-11-10:
Well yes, this issue #861 was also mine, but it was closed, and Milos mentioned, that it will be soved with new build. Situation is that I still have this issue with beta 5.

What do you mean by Bad TBR? TBR I can change in any time. I can also change Basic BR in profile setting. Issue is only when AAPS decide to stop TBR and use basic basal value. In such case it is showen in OVERVIEW ta as TBR 0 U/h with this icon:

[image]

and should be:

[image]

Question is also what is then value taken to IOB calculation.

. In Dash tab it is ok. There is nothing in TBR, and Basic Basal Rate is correct. I guess it is just matter of display in Overview screen. In NS ratio is correct - Basic Basal Rate.

Take a log on attached log.

[AndroidAPS (5).log](https://github.com/nightscout/AndroidAPS/files/7508730/AndroidAPS.5.log)

Acording AAPS Overiev screen Basal was 0 U\h since more or less 00:30 CET.

[image]

But in log you can see that it was set to 0.25 U/h like in Profile setting.

I also have Logs from 6th of Nov, but I don't know if there is a reason to send it.

Thank you

betees82, 2021-11-10:
Ok, this buggy TBR your mean this from SMS? It looks like 2 more hours to go. We will see
[image]

avereha, 2021-11-10:
Yes, this is the TBR I mentioned!
If you still have them, please attach the logs for the last 3/4 days(since the 6th of November). This was the date when this TBR was set.
I'm also interested in the pod history screenshots if you can share them. If there are errors, please scroll down to the first line that doesn't have an error. If all lines have errors, please share the first line in history.

betees82, 2021-11-10:
YES! Problem despaired after this wired TBR 4800 minutes ended. Do you still need logs mentioned above?

betees82, 2021-11-10:
[drive-download-20211110T120347Z-001.zip](https://github.com/nightscout/AndroidAPS/files/7512473/drive-download-20211110T120347Z-001.zip)
all Logs

avereha, 2021-11-10:
Do you have earlier logs?
I see that this TBR was set on 6th at ` @20:43`
The logs start at `21:09` :(

betees82, 2021-11-10:
No, I think I installed new Beta that day, and older logs disappeared. But i can share NS history:

[image]

kristinsmotz, 2021-11-11:
After conferring with @avereha over in Discord, I will add to this issue rather than creating my own. 
History:
Just switched pod and decided to try an Eros pod with beta5. First had to "deactivate" old pod for some reason, even though I deactivated dash pod before AND deactivation of the last Eros pod we had in 2.8.2 was successful 3 weeks ago before we started the beta testing. Then had several instances of system message AAPS is not responding, wait or wait, during the new pod activation. I started screen recording after the first one. Two or three restarts later pod was activating, but the on-screen guidance got stuck. I.e I hit next to insert cannula, confirmed the "are you sure" pop-up, pod started the insertion process, phone stayed in the "hit next to insert cannula" screen, then after cannula inserted I pressed next again, then circle for a fraction of a second, then checkmark pod inserted.
Screen recordings from pod change:
https://user-images.githubusercontent.com/71758682/141377990-ea5c40e6-1415-4c6a-9f95-7f04dba607dd.mp4

https://user-images.githubusercontent.com/71758682/141378003-afa7906a-9a8b-4f17-9afc-4973f1d7171c.mp4

After the pod change I had to bolus for her evening meal, serve it. After that I noticed a 0 basal still running, and on a whim went into pod history. It had a bunch of records for "cancel fake temporary basal created because pod was suspended". 

[image]

[image]

I've since tried canceling the 0 basal several times, but it seems the message isn't coming through to the pod or isn't accepted by pod. I have tried exiting AAPS, suspending and resuming delivery, shutting down phone and restarting. But it keeps apparently not being able to suspend temp basal. New iterations of the "cancel fake temp basal" messages are writing to the pod history every minute. 
Screen recording from attempt to cancel temp basal:
https://user-images.githubusercontent.com/71758682/141378109-7420a95c-c452-4083-b0e1-1c8c11516dce.mp4

We did just hear a "tick" indicating basal is running. 
[image]

---
After conferring with avereha I suggested to try to export settings, force stopping AAPS, deleting all app data from phone, uninstalling and reinstalling/importing settings. And that actually worked. I did not have to kill the pod after all. I will of course keep checking frequently that everything seem well for an hour or two before going to bed. 

I suspect the "bad TBR" that wouldn't go away was the one "created" after switching to Eros in the config builder. I had already deactivated the Dash pod a good half hour before switching to eros, so a "real" 0 TBR from deactivating that pod should have been running. 

[AndroidAPS_LOG_1636664924724.log.zip](https://github.com/nightscout/AndroidAPS/files/7523619/AndroidAPS_LOG_1636664924724.log.zip)

These logs should include the pod change and attempts to cancel fake temp basal, I sent them to myself just after discovering we had a problem.

avereha, 2021-11-12:
Should be fixed with: https://github.com/nightscout/AndroidAPS/commit/85178ee19ff628dfe4422409973c6f8ae7b511b5

avereha, 2021-11-24:
This issue can be closed

closed 2021-11-24 by MilosKozak