nightscout/AndroidAPS #896: Dana-I pairing

MilosKozak, 2021-11-10:
When DANA-i and AAPS first pair, AAPS needs to remember the initial (6 digit) pairing key.  I suspect that currently AAPS is using new pairing key sent each time AAPS and DANA-I communicate?

When not using the initial pair key – when you change a battery within the DANA-i, AAPS looses pairing, and is unable to deliver or temp basal!

As I understand, after the battery change the DANA-I will communicate with a new key…

heikeov, 2021-11-10:
After each battery change I have to repair my Dana-i with a new key.
Most of time I deleted my Dana-i from bluetooth and paired it again, because this seemed to work faster.

MilosKozak, 2021-11-10:
fixed by https://github.com/nightscout/AndroidAPS/commit/7ad425ae1d49f86b0e177e222cf4a802a64af984

closed 2021-11-10 by MilosKozak