nightscout/AndroidAPS #897: (3.0beta5) Frozen screen/app crashes/update basal profile error

Els3401, 2021-11-10:
Aaps 3.0 beta 5
Combo/Ruffy
Dexcom G6 (Build your own app)
Phone: Google Pixel 4XL Android 12

Problems:
- Screen freezes sometimes when I switch profile. Sometimes the app crashes with the message wait or close. I get that message a lot more now. 
- The connection with combo looks slower. 
- Update of basal profile failed error (at night, sometimes during the day
Thanks a lot.

Battery optimization is off, Play Protect is off.
The app has all rights, ...
[image]
[image]
[image]
[image]
[AndroidAPS._2021-11-10_06-20-19_.19.zip](https://github.com/nightscout/AndroidAPS/files/7512549/AndroidAPS._2021-11-10_06-20-19_.19.zip)

MilosKozak, 2021-11-10:
that's not enough info to resolve the issue
do you have logfiles?

Els3401, 2021-11-10:
[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/7513877/AndroidAPS.log)
Excuse, I submitted the wrong file.

Alamo04, 2021-11-10:
Confirmed for Pixel 5, it comes with Android 12, and all Defs, Betas are involved.
I think its not AAPS, but Android 12 has still some problems.

MilosKozak, 2021-11-10:
there is 6 min of log files and nothing supsicious there

Els3401, 2021-11-11:
[AndroidAPS._2021-11-10_06-20-19_.19 2 kopie.txt](https://github.com/nightscout/AndroidAPS/files/7519658/AndroidAPS._2021-11-10_06-20-19_.19.2.kopie.txt)
I hope that this file can help. The error happened around 11:25.
I switched again to master yesterday. And I find only zips of yesterday afternoon :s.

MilosKozak, 2021-11-11:
i believe we need even following log
@jotomo ?

jotomo, 2021-11-11:
At timestamp 11:30:12.011 the combo stops responding to key presses. Connection is there, pump display is returned, but no response to key presses. Pump then returns to main screen due to timeout.
@dv1: could the comm buffer be full? Setting a new basal profile is the longest running command with a crazy amount of key presses.

dv1, 2021-11-11:
@jotomo We'd need Ruffy logs to be sure. These should show error codes in the app layer packets that have a reliability bit set.

Els3401, 2021-11-11:
Where can I find those Ruffy logs?

dv1, 2021-11-11:
That's where it gets more complicated, since Ruffy uses Android.Log - it does not write to a text file. You'd have to use `adb logcat`, but this captures _all_ logging from Android and Android apps, so you'd have to filter out the Ruffy bits.

Els3401, 2021-11-11:
I find only new logs. I did a profile switch. 
[Ruffy.txt](https://github.com/nightscout/AndroidAPS/files/7521644/Ruffy.txt)

Combo log during a profile switch
2021-11-11 16:54:38.477 12047-4449/? D/PUMP: [Thread-1377]: [ComboPlugin.disconnect():244]: Disconnect called with reason: Queue empty
2021-11-11 16:54:39.755 12047-12175/? D/AUTOMATION: [RxCachedThreadScheduler-13]: [AutomationPlugin$onStart$13.accept():142]: Grabbed new BT event: EventBTChange[deviceAddress=00:0E:2F:E2:EF:D9,deviceName=SpiritCombo,state=DISCONNECT]
2021-11-11 16:56:04.341 12047-12175/? D/AUTOMATION: [RxCachedThreadScheduler-13]: [AutomationPlugin$onStart$13.accept():142]: Grabbed new BT event: EventBTChange[deviceAddress=00:0E:2F:E2:EF:D9,deviceName=SpiritCombo,state=CONNECT]
2021-11-11 16:56:06.306 12047-5195/? D/PUMP: [Thread-1461]: [ComboPlugin.checkHistory():1216]: Setting 'pumpHistoryChanged' false
2021-11-11 16:57:52.630 12047-5195/? D/PUMP: [Thread-1461]: [ComboPlugin.disconnect():244]: Disconnect called with reason: Queue empty
2021-11-11 16:57:54.035 12047-12175/? D/AUTOMATION: [RxCachedThreadScheduler-13]: [AutomationPlugin$onStart$13.accept():142]: Grabbed new BT event: EventBTChange[deviceAddress=00:0E:2F:E2:EF:D9,deviceName=SpiritCombo,state=DISCONNECT]
2021-11-11 16:59:54.177 12047-12175/? D/AUTOMATION: [RxCachedThreadScheduler-13]: [AutomationPlugin$onStart$13.accept():142]: Grabbed new BT event: EventBTChange[deviceAddress=00:0E:2F:E2:EF:D9,deviceName=SpiritCombo,state=CONNECT]
2021-11-11 16:59:56.689 12047-5754/? D/PUMP: [Thread-1500]: [ComboPlugin.checkHistory():1216]: Setting 'pumpHistoryChanged' false
2021-11-11 17:01:53.106 12047-5754/? D/PUMP: [Thread-1500]: [ComboPlugin.disconnect():244]: Disconnect called with reason: Queue empty
2021-11-11 17:01:54.409 12047-12175/? D/AUTOMATION: [RxCachedThreadScheduler-13]: [AutomationPlugin$onStart$13.accept():142]: Grabbed new BT event: EventBTChange[deviceAddress=00:0E:2F:E2:EF:D9,deviceName=SpiritCombo,state=DISCONNECT]

jotomo, 2021-11-13:
@Els3401 
1) Thanks for the ruffy.log file. In that instance the basal profile was updated successfully, correct? We'd need a log showing an unsuccessful attempt to figure out what's going wrong.
2) When you say you see more errors now, when did you see fewer errors, what has changed about your setup? Update to Android 12, new phone, update to AAPS 3 beta?

Els3401, 2021-11-13:
Thats correct. I transfered to master again because of the problems. So, I can't give you a log. Now I get the error only when I am to far away.
When I have changed to beta 5, I saw the error a lot, freezing of the screen. I started with beta 4, but soon there was beta 5. 
I had also problems at night, when phone and combo are close together.
In master I have no problems with android 12 and aaps works fine (only the triangles). 
If you want me to transfer again to beta 5... I will do that and try to log ruffy. not simply because the log is very short and the phone has to be connected with laptop.

Els3401, 2021-11-28:
[Ruffy log.txt](https://github.com/nightscout/AndroidAPS/files/7613174/Ruffy.log.txt)

The problems occur when there is a profile switch. Or sometimes at night.
Beta 7

dv1, 2021-11-28:
Hmm unfortunately I do not see anything that hints at the underlying cause. I do see the error, but not any more details. The RT_DISPLAY, RT_AUDIO etc. packets just eventually stop coming in because for unknown reasons the BT connection fails. I also checked the application layer packets that have the reliability bit set, and their error codes are set to 0. A nonzero code would give us more details.

Still, my main suspicion is also that the ringbuffer inside the Combo overruns. I see the following two options for how to proceed:

1. Use git bisect to see if you can pinpoint a particular commit that introduced this problem.
2. Lengthen the waiting periods between the basal rate factor updates (and maybe also during the factor updates), then see if this still happens. This amounts to tweaking some of the sleep() calls inside the Combo AAPS plugin. @jotomo , any suggestions which ones to tweak?

jotomo, 2021-11-28:
There's an additional sleep here (in addition the to 150ms after each press): https://github.com/nightscout/AndroidAPS/blob/32873e073087e8dc05c483d29bb836885b77e66c/combo/src/main/java/info/nightscout/androidaps/plugins/pump/combo/ruffyscripter/commands/SetBasalProfileCommand.java#L100
Though that code hasn't changed in four years.
I went through the changes between beta3 and beta5 (https://github.com/nightscout/AndroidAPS/compare/3.0-beta3...3.0-beta5), but there's nothing suspicious there (no change in Combo code, nothing that could cause hangs/ANRs).
I doubt the issue is in ComboPlugin or ruffyscripter-land, the code is mature and functionally untouched for quiet a while.
... somewhat at a loss as to what might cause the observed behavior.

dv1, 2021-11-29:
@jotomo  Perhaps this is a dormant bug that is only triggered in Android 12 for some reason. It would perhaps be useful to try to see if this happens in Android 10 and 11 as well.

dv1, 2021-11-30:
@Els3401 Can you try to raise the sleep period in the location @jotomo mentioned from 50 to 200 for example? Perhaps it works then for you?

Els3401, 2021-12-06:
Thanks a lot. I have done this change to 200 several days ago and build a new apk. I don't see any changes :(. Or do I something wrong? I have also tried 250.
I have done also a reset of wifi, bluetooth and mobile. I deleted Aaps and Ruffy before (and deleted also the bluetooth connections). And installed everything following the docs. Sadly, it doesn't help. 
Return to Android 11 seems very difficult.
I use beta 9.

Els3401, 2021-12-14:
I switched from phone Monday evening (android 11).
I got today (Tuesday) 2 failed update of basal profile. Bolus looks fine, connection also. I use UAM and have automations with profile switches. The error occurs then.
No freezes anymore :).
Build: 6e290a631-2021.12.02-13:33
Flavor: full release
Engineering mode is activated
AndroidAps 3.0 - beta 9
Phone: Samsung S21 Ultra
[AndroidAPS._2021-12-14_06-44-10_.39.zip](https://github.com/nightscout/AndroidAPS/files/7714297/AndroidAPS._2021-12-14_06-44-10_.39.zip)
[AndroidAPS._2021-12-14_17-09-03_.0.zip](https://github.com/nightscout/AndroidAPS/files/7714301/AndroidAPS._2021-12-14_17-09-03_.0.zip)

closed 2022-05-23 by MilosKozak