nightscout/AndroidAPS #899: Crash on changing Overview graph granularity

rustymonkey, 2021-11-10:
When changing the granularity of the graph on the Overview tab, AAPS crashes.

0 AAPS is looping / graph is set to display 'basals' and 'activity' with no other graphs (graph1, graph2,...) enabled
1 Tap-hold on the graph to change granularity
--> after performing step 1 two or three times, AAPS crashes 

Note that the issue can be consistently replicated.
--------------
Tested on AAPS 3.0 B5, running on an S7/Android 11 (Pixel Experience) connected to MDT 554 via RileyLink
Logs shared directly with Milos. Logs were generated after restarting AAPS immediately after the crash occurred.

MilosKozak, 2021-11-10:
i improved UI much after beta5 and now i cannot replicate. but it can have another reason.
update this issue after you test it with beta6 (when released) or current dev

rustymonkey, 2021-11-11:
Tested on 2.8.2.10-dev build 2021-11-11 11:40 and the issues are partially resolved (it no longer crashes). Tap-hold once always works as expected. But tap-hold 2 or 3 times results in:
1) the UI doesn't respond immediately to the 2nd/3rd tap-hold (always)
2) the Android dialog 'AndroidAPS isn't responding' appears (often)

From an end user perspective, it seems as if AAPS is doing something CPU-intensive after switching granularity once, and attempting to switch again while that background activity is going on doesn't put a stop to that activity, resulting in the the UI being non-responsive.

I've shared new logs illustrating this with the above build.

MilosKozak, 2021-11-12:
well if you change scale data in the past must be calcalculated
on fast phone it's under 10s of 100% utilization
on slow it's tens of seconds

if you press it more times recalculation is restarted which takes even more time
i optimized the calculations many times and i don't see any space for futher speedup
so. if if you are not happy with current state just buy faster phone

closed 2021-11-12 by MilosKozak

Philoul, 2021-11-12:
This issue is similar to #801 (I never had crash but very often App not responding)
It's a bit better on latest dev, but I still have App Not responding popup and the same behaviour explained by @rustymonkey in his last comment

Philoul, 2021-11-12:
> well if you change scale data in the past must be calcalculated
on fast phone it's under 10s of 100% utilization
on slow it's tens of seconds

if you press it more times recalculation is restarted which takes even more time
i optimized the calculations many times and i don't see any space for futher speedup
so. if if you are not happy with current state just buy faster phone

I could improved it with doing during prepare data all scaling possibilities for the last 24hours (6h scaling, 12h scaling, 18hour scaling, 24 hour scaling, with and without prediction)

 It worked fine in overview because no additional calculation was required during any change of scaling done in overview (I saw immediatly all curves). I closed my PR because not ok with HistoryBrowser 😞.
I can update and propose again this PR as starting point.

Philoul, 2021-11-12:
@rustymonkey can you test #911 and tell me if it works better ?

rustymonkey, 2021-11-14:
Tested this PR on 2.8.2.10dev 2021-11-14-12:49, and it resolves the issue. i.e.
- the UI now responds consistently to Tap-hold actions
- tap-hold actions no longer trigger the 'AndroidAPS isn't responding' dialog

rustymonkey, 2022-01-17:
Note: My previous comment might be misleading--PR #911 isn't yet included with AAPS 3.0. It's a work in progress, tracked at #801.