nightscout/AndroidAPS #900: Medtronic tab initially renders the careportal [refresh problem]

rustymonkey, 2021-11-10:
When importing settings into AAPS 3.0, the pump tab initially renders the old careportal (or some variant thereof). The workaround to this is to either:
a) reconfigure the pump
b) in the Profile builder, disable visibility for the Medtronic tab, and then re-enable it
[image]

Tested AAPS 3.0 B5

MilosKozak, 2021-11-11:
is the app restarted aftr you import?

MilosKozak, 2021-11-17:
it happened to me too but currently i don't have solution. restart helps

MilosKozak, 2023-08-17:
Closed for inactivity. Create new issue is if you still see the problem on latest branch

closed 2023-08-17 by MilosKozak