nightscout/AndroidAPS #901: [3.0-Beta5] Screen freezes when initializing Omnipod Eros

regiorunner, 2021-11-10:
Phone: Moto 5 G5 Plus 6GB / 128GB
Android 11
3.0-Beta5

After succesful trial with Omnipod Dash switched back to Omnipod Eros.
After making the necessary changes and coupling the Rileylink prepared the Eros pod.
Entering "Activate Pod" and pressing in the bottom on "next" nothing seemed to happen.
After some time Android showed a message "APS does not seem to react. Close or wait?" I pressed "wait".
At that moment the Eros started clicking, indicating the initialization procedure was started.
When the clicking stopped (indicating initialization is finished) the correct screen appeared.
Pressing next to insert the canula the pop-up screen appears. Pressing "OK" freezes again the screen and again Android
showed the message  "APS does not seem to react. Close or wait?" Again pressed "wait".
After a few seconds the Pod starts clicking and the canula was inserted. The command was send correct to the Eros pod. After completion the correct screen appeared.
In the Pod tab pressing "updating status" works immediately, indicating connection to Eros pod is ok. Also injecting insuline and automatic basal aaptation works ok.
Did not encounter these problems initializing the Omnipod Dash, which uses simular screens.

If more info is needed please let me know. I do have the logfiles of course.

betees82, 2021-11-10:
Same Issue with Eros. There's now turning blue wheel during steps. 
[image]

StephenBrown2, 2021-11-10:
Duplicate of https://github.com/nightscout/AndroidAPS/issues/782?

betees82, 2021-11-10:
> Duplicate of #782?

Looks like dublicate

closed 2021-11-11 by MilosKozak

regiorunner, 2021-11-11:
It indeed looks like a duplicate of #782. Thanks for your comments.