nightscout/AndroidAPS #903: No way to acknowledge persistent alarm - Dash

ToreBj, 2021-11-11:
I had my Dash pod running for the extended period (8h) until it gave a continuous beeping tone, i.e. expired.
I deactivated the pod, everything was reset and ready for init of a new pod.
I activated the new pod with no problems.
However now I started to hear an increasing alarm sound, but could not see any notification to acknowledge.

I did the Dash pod change at 17:46. 10-nov-2021.

Restarting AAPS made no difference. I finally force stopped AAPS and the alarm was reset.

 
[image]

[AndroidAPS.log](https://github.com/nightscout/AndroidAPS/files/7518947/AndroidAPS.log)

I'm using a Oneplus 7 Pro with Android v10. 
Dexcom G6 patched app, Xdrip+

kristinsmotz, 2021-11-11:
Sounds similar to what I had in this issue: https://github.com/nightscout/AndroidAPS/issues/816
In that case the old pod failed during deactivarion, and a two-tone alert kept sounding. I tried to exit app, but no help, so finished activating new pod then shut down the phone to get rid of it. I think I saw a "null" error message in system notification at some point, but swiping it didn't kill the alert.

avereha, 2021-11-15:
Fixed with https://github.com/nightscout/AndroidAPS/pull/916

closed 2021-11-16 by MilosKozak