nightscout/AndroidAPS #904: [2.8.2.10-dev] Wrong Basal IOB in CALCULATOR Button (if Extended bolus is running ?)

Philoul, 2021-11-11:
For my lunch I first opened twice Calculator (and cancel dialog) and I was surprised to see very different calculation for proposed bolus, so I took a series of screenshots (max 2 minutes between the fist screenshot and the last one) from the upper left to the lower right.

Test done this way:
=> Open Calculator, take the screenshot, cancel Calculator about 10 times, and one screenshot of detailled IOB to get real value

Note: An extended bolus was running during these screenshots (from 14:08 to 14:12). I suppose it could be the reason of the bug because I made the same verification one hour later without extended bolus and everything was OK, but I'm not sure...

see below results: all basal IOB are different, and not consistent with first screenshot and of course results of wizard calculator are very different...
[image]

I look into log file but I found nothing interesting inside it (all dialogs where cancelled so I don't see detailled calculation of basal IOB)
[AndroidAPS._2021-11-11_00-00-03_.17.zip](https://github.com/nightscout/AndroidAPS/files/7521049/AndroidAPS._2021-11-11_00-00-03_.17.zip)

MilosKozak, 2021-11-11:
i have the results absolutely consistent

MilosKozak, 2021-11-11:
if you can replicate the issue add more logging to `IobCobCalculator::calculateIobToTimeFromTempBasalsIncludingConvertedExtended`
ie # of records first, later records to find the difference

Philoul, 2021-11-11:
Later in the afternoon (without extended tbr) it was perfect, and then again during another extended bolus running, variation of basal IOB  and some inconsistancies... (not sure it's link to extended bolus but it seems to be the case).
when I run several times (quickly) Calculator and Cancel, when it's perfect I always get same basal IOB (or only updated with -0.01 or -0.02 U). When the problem occurs, on each new Dialog, basal IOB is different (-0.2U, -0.4U, +0.1U...) and change in both direction (increase or decrease).
I will try to add logs in BolusWizard and in iobCobCalculator to understand why...

Philoul, 2021-11-11:
I included some logs (see screenshot below, and I confirm there is an issue in IOB calculation for the contribution of a running Extended TBR (see global curves below):
Here the results of 3min30s of Basal IOB calculation (first point at 18h45min41s and last point at 18h49min24s)
An extended bolus was running from 18h42 to 18h48
A 0 TBR started after the Extended Bolus
In blue contribution of Basal to IOB, in Red contribution of all Extended Bolus and in green result (Total Basal IOB)...
[image]

Below you see the contribution of running extended TBR in blue (values are wrong during the run and starts to be Ok after the end)
I included the contribution of a previous Extended Bolus in red to compare
[image]

Not very "academic" but see below the log added
[image]

and extract of log of calculated IOB for the Extended TBR that started at 18h42
```
18:47:23.301 [main] D/CORE: [IobCobCalculatorPlugin.calculateIobToTimeFromTempBasalsIncludingConvertedExtended():637]: XXXX Ext11/11/2021 18:42:46 >>> 0.8446998888311451
18:47:23.505 [main] D/CORE: [IobCobCalculatorPlugin.calculateIobToTimeFromTempBasalsIncludingConvertedExtended():637]: XXXX Ext11/11/2021 18:42:46 >>> 0.8441400119682051
18:47:26.919 [main] D/CORE: [IobCobCalculatorPlugin.calculateIobToTimeFromTempBasalsIncludingConvertedExtended():637]: XXXX Ext11/11/2021 18:42:46 >>> 0.8337204659186268
18:47:57.543 [main] D/CORE: [IobCobCalculatorPlugin.calculateIobToTimeFromTempBasalsIncludingConvertedExtended():637]: XXXX Ext11/11/2021 18:42:46 >>> 0.7509720444581759
18:47:57.714 [main] D/CORE: [IobCobCalculatorPlugin.calculateIobToTimeFromTempBasalsIncludingConvertedExtended():637]: XXXX Ext11/11/2021 18:42:46 >>> 0.750569127598282
18:48:00.056 [main] D/CORE: [IobCobCalculatorPlugin.calculateIobToTimeFromTempBasalsIncludingConvertedExtended():637]: XXXX Ext11/11/2021 18:42:46 >>> 0.7447512375450931
18:48:00.141 [main] D/CORE: [IobCobCalculatorPlugin.calculateIobToTimeFromTempBasalsIncludingConvertedExtended():637]: XXXX Ext11/11/2021 18:42:46 >>> 0.7445541843904657
18:48:00.389 [main] D/CORE: [IobCobCalculatorPlugin.calculateIobToTimeFromTempBasalsIncludingConvertedExtended():637]: XXXX Ext11/11/2021 18:42:46 >>> 0.7440164130281922
18:48:04.538 [main] D/CORE: [IobCobCalculatorPlugin.calculateIobToTimeFromTempBasalsIncludingConvertedExtended():637]: XXXX Ext11/11/2021 18:42:46 >>> 0.7341119383229607
18:48:18.798 [main] D/CORE: [IobCobCalculatorPlugin.calculateIobToTimeFromTempBasalsIncludingConvertedExtended():637]: XXXX Ext11/11/2021 18:42:46 >>> 0.8429244569179892
18:48:18.941 [main] D/CORE: [IobCobCalculatorPlugin.calculateIobToTimeFromTempBasalsIncludingConvertedExtended():637]: XXXX Ext11/11/2021 18:42:46 >>> 0.8425602082927663
18:48:19.224 [main] D/CORE: [IobCobCalculatorPlugin.calculateIobToTimeFromTempBasalsIncludingConvertedExtended():637]: XXXX Ext11/11/2021 18:42:46 >>> 0.8418761516435282
18:48:22.110 [main] D/CORE: [IobCobCalculatorPlugin.calculateIobToTimeFromTempBasalsIncludingConvertedExtended():637]: XXXX Ext11/11/2021 18:42:46 >>> 0.8344905580709777
18:48:36.603 [main] D/CORE: [IobCobCalculatorPlugin.calculateIobToTimeFromTempBasalsIncludingConvertedExtended():637]: XXXX Ext11/11/2021 18:42:46 >>> 0.1702371645821504
18:48:36.738 [main] D/CORE: [IobCobCalculatorPlugin.calculateIobToTimeFromTempBasalsIncludingConvertedExtended():637]: XXXX Ext11/11/2021 18:42:46 >>> 0.17023639321626927
18:48:40.510 [main] D/CORE: [IobCobCalculatorPlugin.calculateIobToTimeFromTempBasalsIncludingConvertedExtended():637]: XXXX Ext11/11/2021 18:42:46 >>> 0.17021288280759417
18:48:40.599 [main] D/CORE: [IobCobCalculatorPlugin.calculateIobToTimeFromTempBasalsIncludingConvertedExtended():637]: XXXX Ext11/11/2021 18:42:46 >>> 0.17021225733992026
18:48:40.816 [main] D/CORE: [IobCobCalculatorPlugin.calculateIobToTimeFromTempBasalsIncludingConvertedExtended():637]: XXXX Ext11/11/2021 18:42:46 >>> 0.17021098659919603

```

MilosKozak, 2021-11-11:
great work @Philoul !!
it was not one bug but two :(
fixed by https://github.com/nightscout/AndroidAPS/commit/8bac46830641a36934813d16b98d5413723baff3

closed 2021-11-11 by MilosKozak