nightscout/AndroidAPS #910: AAPS shutting down and disconnecting from pump everyday at 3AM after 2.8.2.1 update.

GrapesGithub, 2021-11-12:
**Problem:** 
I have 2 kids on AndroidAPS and updated both from 2.6.1.2 to 2.8.2.1 a few weeks ago. After the update, only in 1 of the kids (the other one is fine… weird), I notice that almost every night, at around 3AM the AAPS disconnects from the pump and so it stays in open loop. I only figure it out either when I normally wake up or when her glucose gets out of the limits I set on Dexcom.

**My system**: 
Xiaomi 4x with android 8, accuchek combo and a Dexcom, and I get the glucose readings into the app through nightscout (not the Dexcom patched app)

**What I already checked/tried:** 
1) both AAPS and Ruffy are NOT on battery optimization. 
2) I left the phone charging all night and I still got the issue.

**Attached files:**
I attach hee logs of before the issue and recent ones. 
I also attach photos of how AAPS looks like as soon as I realize the issue, what I then do is refresh accuchek and restart the Nsclient and all goes back to normal.

Can someone help? Is there any detail you need from me to find the issue? I am out of ideas and I have no clue how to interpret the logs…

[image]
[image]
[image]
[logsaaps.zip](https://github.com/nightscout/AndroidAPS/files/7526304/logsaaps.zip)

MilosKozak, 2021-11-12:
disable play protect on your phone

closed 2021-11-12 by MilosKozak

GrapesGithub, 2021-11-12:
Aha! Thank you. 
The play protect scanning time actualy matches the time of shutting down. I have disconnected and will revert back if it solved the issue. Thank you again

GrapesGithub, 2021-11-14:
It's been 2 nights without the issue, so i guess it really was the play protect thing. Thank you so much for the help.