nightscout/AndroidAPS #917: [3.0-beta5] Problem with temporary basals

fjpezuela, 2021-11-13:
AAPS 3.0-beta5

On many occasions, the calculation that he makes of the temporary basal is immensely high.

I think this problem may be the one causing the problem in the statistics that I recently reported in the following ticket: https://github.com/nightscout/AndroidAPS/issues/859

I add a screenshot where you can see the problem:

[image]

Regards,

Fran.

andyrozman, 2021-11-14:
This seems like problem with your setup (NS).

Marked entry looks like some sort of daily total... 55 U, looks like your daily dose... 

Upper 3 entries all look like they are happening at same time... 

On which pump does this happen?

fjpezuela, 2021-11-15:
Hello,

I am using Accu-Check Insight.

This problem occurs in both version 3.0-beta5 and dev version 2.8.2.10-dev. Yesterday I updated to the dev version in case the problem had been corrected, I reset the database and I have seen that the same thing continues to happen.

In stable version 2.8.2 it has never happened to me, only in betas and dev.

My basals are in this range: 0.27-0.45 and my multipliers in AAPS are x3 and x4, so I don't understand how I can get to those values.

I leave some more captures and logs:

[image]

[image]

[image]

[AndroidAPS._2021-11-14_00-00-00_.4.zip](https://github.com/nightscout/AndroidAPS/files/7537517/AndroidAPS._2021-11-14_00-00-00_.4.zip)
[AndroidAPS._2021-11-15_00-00-04_.0.zip](https://github.com/nightscout/AndroidAPS/files/7537519/AndroidAPS._2021-11-15_00-00-04_.0.zip)
[AndroidAPS._2021-11-15_08-15-17_.0.zip](https://github.com/nightscout/AndroidAPS/files/7537520/AndroidAPS._2021-11-15_08-15-17_.0.zip)

Philoul, 2021-11-16:
@fjpezuela can you test again with beta6 please?
Should be fixed now...

closed 2021-11-17 by MilosKozak