nightscout/AndroidAPS #918: [3.0-beta5] Error while reading history of DanaRS

Snoopy913, 2021-11-13:
Version: AndroidAPS 3.0-beta5
Pumpe: DanaRS v3
                Model: 05
                Protokoll: 11
                Code: 11

I have the issue that often, when a connection to the pump happens, the connection is not ended for nearly a minute.

In the logs I saw that when the pump history is requested, after receiving all the history events, there is a Java error shown:

```
12:55:12.238 [Binder:20818_3] D/PUMPCOMM: [DanaRSPacketAPSHistoryEvents.handleMessage():77]: Last record received
12:55:12.274 [RxCachedThreadScheduler-79] E/DATABASE: [PumpSyncImplementation.syncStopTemporaryBasalWithPumpId$lambda-21():269]: Error while saving TemporaryBasal
java.lang.IllegalArgumentException: Failed requirement.
	at info.nightscout.androidaps.database.interfaces.DBEntryWithTimeAndDurationKt.setEnd(DBEntryWithTimeAndDuration.kt:11) ~[na:0.0]
	at info.nightscout.androidaps.database.transactions.SyncPumpCancelTemporaryBasalIfAnyTransaction.run$database_fullRelease(SyncPumpCancelTemporaryBasalIfAnyTransaction.kt:18) ~[na:0.0]
	at info.nightscout.androidaps.database.transactions.SyncPumpCancelTemporaryBasalIfAnyTransaction.run$database_fullRelease(SyncPumpCancelTemporaryBasalIfAnyTransaction.kt:7) ~[na:0.0]
	at info.nightscout.androidaps.database.AppRepository.runTransactionForResult$lambda-4$lambda-3(AppRepository.kt:53) ~[na:0.0]
	at info.nightscout.androidaps.database.AppRepository.$r8$lambda$PutkuiY9RpEna-m2EmcDL4xFiLQ(Unknown Source:0) ~[na:0.0]
	at info.nightscout.androidaps.database.AppRepository$$ExternalSyntheticLambda71.call(Unknown Source:6) ~[na:0.0]
	at androidx.room.RoomDatabase.runInTransaction(RoomDatabase.java:577) ~[na:0.0]
	at info.nightscout.androidaps.database.AppRepository.runTransactionForResult$lambda-4(AppRepository.kt:51) ~[na:0.0]
	at info.nightscout.androidaps.database.AppRepository.$r8$lambda$S47IsDF-EPWuM90x6Kca_DdYKHM(Unknown Source:0) ~[na:0.0]
	at info.nightscout.androidaps.database.AppRepository$$ExternalSyntheticLambda69.call(Unknown Source:6) ~[na:0.0]
	at io.reactivex.internal.operators.single.SingleFromCallable.subscribeActual(SingleFromCallable.java:44) ~[na:0.0]
	at io.reactivex.Single.subscribe(Single.java:3666) ~[na:0.0]
	at io.reactivex.internal.operators.single.SingleSubscribeOn$SubscribeOnObserver.run(SingleSubscribeOn.java:89) ~[na:0.0]
	at io.reactivex.Scheduler$DisposeTask.run(Scheduler.java:608) ~[na:0.0]
	at io.reactivex.internal.schedulers.ScheduledRunnable.run(ScheduledRunnable.java:66) ~[na:0.0]
	at io.reactivex.internal.schedulers.ScheduledRunnable.call(ScheduledRunnable.java:57) ~[na:0.0]
	at java.util.concurrent.FutureTask.run(FutureTask.java:266) ~[na:0.0]
	at java.util.concurrent.ScheduledThreadPoolExecutor$ScheduledFutureTask.run(ScheduledThreadPoolExecutor.java:301) ~[na:0.0]
	at java.util.concurrent.ThreadPoolExecutor.runWorker(ThreadPoolExecutor.java:1167) ~[na:0.0]
	at java.util.concurrent.ThreadPoolExecutor$Worker.run(ThreadPoolExecutor.java:641) ~[na:0.0]
	at java.lang.Thread.run(Thread.java:923) ~[na:0.0]
```

It looks like AAPS has troubles with some history events.

Maybe someone can interpret the error message. If additional info is needed I will try to provide it.

MilosKozak, 2021-11-15:
I need to turn on all PUMP logging and provide full log (starting at least 10 min before) when this happen again

Snoopy913, 2021-11-15:
@MilosKozak 
PUMP loggings where on, starting at 01:57 it occured several times.
 I hope it helps:
[AndroidAPS._2021-11-13_03-22-53_.txt](https://github.com/nightscout/AndroidAPS/files/7538795/AndroidAPS._2021-11-13_03-22-53_.txt)

Since the last error occurance in this logfile, it didn't occure again.

MilosKozak, 2021-11-17:
your pump doesn't behave correctly (not all the time).
for example for sequence STOP TBR , START NEW TBR responds
STOP FAILED
START OK
is TBR running = false

and in history there are both records (for START & STOP)

i'd recommend reset pump and/or replace if needed

closed 2021-11-17 by MilosKozak